botletics/SIM7000-LTE-Shield #91: Connection issue

rajendran20, 2019-05-03:
Currently I am using uno with the botlectics shield with my IoT sim card (verified working). However there seems to be an issue when running LTE_demo code. I cant seem to register to the network. (status is 0). I am using USB to laptop with external power to uno. 

When I plugged into the micro usb port of the shield to check AT commands using Putty, the shield comm ports becomes unstable (keeps going on and off). Not sure is it due to both uno and shield are connected to the laptop. Tried to use only the 9V external power to uno but still same.

However when I use a uno which is not able to be detected by the laptop (driver ic spoilt), my port for the shield is stable and I can invoke AT commands and able to confirm network can be registered (status 1).

Any advise on this weird issue?

botletics, 2019-05-06:
Please make sure you have a LiPo battery connected. That's really important for it to work!

rajendran20, 2019-05-06:
Hi Tim,

Will do that, thanks for getting back. One more thing, is there any connection diagram for connecting to ESP32 (nodemcu)? Or just need to connect UART interface?

Regards
Raj

botletics, 2019-05-06:
For ESP32 wiring please see the comments section of [the ESP demo code](https://github.com/botletics/SIM7000-LTE-Shield/tree/master/Code/examples/ESP32_LTE_Demo) and [the pinouts wiki page](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Pinouts)

rajendran20, 2019-05-06:
Hi,

Thanks once again..great help

Regards
Raj

rajendran20, 2019-05-07:
Hi Timothy,

Yes the LiPo battery was the issue. My shield was registered to the network after connecting to the LiPo batt. Afterwhich i downloaded IoT_example.ino for GPS tracking. I did encounter the following error where I encountered connection error.

Do you have any advise for me regarding this?

Found 'eeeeem!
---------------------
Latitude: -0.000000
Longitude: 0.000000
Speed: 0.00
Heading: 0.00
Altitude: 0.00
---------------------
 ---> AT+CIPSTATUS
 <--- OK
 <--- STATE: PDP DEACT
Connecting to MQTT...
 ---> AT+CIPSHUT
 <--- SHUT OK
 ---> AT+CIPMUX=0
 <--- OK
 ---> AT+CIPRXGET=1
 <--- OK
AT+CIPSTART="TCP","io.adafruit.com","1883"
 <--- OK
 <--- STATE: PDP DEACT
Connection failed
Retrying MQTT connection in 5 seconds...
 ---> AT+CIPSTATUS
 <--- OK
 <--- STATE: PDP DEACT
 ---> AT+CIPCLOSE
 <--- ERROR
 ---> AT+CIPSHUT
 <--- SHUT OK
 ---> AT+CIPMUX=0
 <--- OK
 ---> AT+CIPRXGET=1
 <--- OK
AT+CIPSTART="TCP","io.adafruit.com","1883"
 <--- OK
 <--- STATE: PDP DEACT
Connection failed
Retrying MQTT connection in 5 seconds...

Sent from Windows Mail

botletics, 2019-05-13:
Make sure your AdafruitIO credentials are correct on the sketch.

closed 2019-06-10 by botletics