botletics/SIM7000-LTE-Shield #93: IMEI isnt being found

Audax56, 2019-05-10:
[image]

Im using a Sim7000G, the LTE_DEMO worked but the IoT-Example just gets stuck here every time

botletics, 2019-05-13:
Please make sure you have a LiPo battery connected with proper polarity.

closed 2019-06-10 by botletics

sprice-HE, 2019-07-04:
I'm having the same issue with the Sim7000A. The red wire is on the left when the connector is closest to you with the sim card on the left as well. As stated by Audax56, the LTE_DEMO worked great, but the IoT example gets hung up at the IMEI.

Audax56, 2019-07-04:
Try commenting out lines 202, 205 & 304 in IoT_example. That got it working
for me

sprice-HE, 2019-07-04:
@Audax56 Thanks!  Commenting out the lines associated with the temp sensor worked.