botletics/SIM7000-LTE-Shield #96: Sim7000E firmware upgrade

DamienLobb, 2019-05-17:
Hello, 

I recently tried upgrading the firmware on the Sim 7000E from B02 to B07 with the firmware given in this repo. When I query the model using AT+CGMM, it replies with SIM7000A not SIM7000E.
I tried the B05 version in this repo too, and again it's SIM7000A. 

I am missing something here? I'm in the UK so I need the 7000E.

Thanks, 
Damien

botletics, 2019-05-20:
You need to use the Qualcomm QPST tool to back up the file according to [these instructions] (https://github.com/botletics/SIM7000-LTE-Shield/blob/master/SIM7000%20Documentation/Firmware%20Updater%20Tool/How%20to%20Update%20SIM7000%20FW%20version%20with%20QPST.pdf)

DamienLobb, 2019-05-20:
If by that you mean updating the xQCN file, I am a little confused. There is no xQCN file in any of the firmware folders, the only time I can see one is on the backup tab, where it is already filled in with "DEADD00D_0.xqcn"

DamienLobb, 2019-05-20:
Apologies, I assume by backup you mean I should have backed up the firmware that was already on there using a xQCN file? If so the instructions could be a little more clear...

DamienLobb, 2019-05-20:
Issue is fixed. 

I was sent the most up to date firmware by SIM Com. For my SIM7000e (part number S2-10732-Z1T52) the most up to date FW that is compatible is the B06 FW. 

You were correct about the xQCN file, however, there is no such file in the .rar folders for either of the 7000e firmwares.

closed 2019-05-20 by DamienLobb