botletics/SIM7000-LTE-Shield #98: Issue with fixed Data length being sent over http

bihmedical, 2019-05-22:
I have been working on ESP32+SIM700 LTE module.  Everything seems to be working fine, the only issue I see is when the Data length goes over 380 the system crashes. 
I have to send some 1000 bytes of data through the module but this is becoming a limiting factor.  I tried looking at the Adafruit_Fona.cpp library but couldn't find anything specific to Data Length. 

[image]

Any inputs on where to change the max Data length will be helpful.
Thank you.

botletics, 2019-05-26:
Doing large payloads can be risky in terms of crashing your Arduino or microcontroller, but you can try to increase the size of URL. Simply go to the IoT_Demo code and find the line that declares URL[200] and change 200 to 1000 or whatever you want. Note: Arduino Uno won't be able to handle that, try Arduino Mega to start off with!

XXplode, 2019-06-08:
Have you had any success with this?

It's it possible for you to send 250 bytes x4 times to keep the memory requirements down?

bihmedical, 2019-06-10:
Thank you for both the responses. I got diverted to another project. I will try and implement these changes and see if it works for me. I will give an update based on the results.

bihmedical, 2019-06-13:
Hello,
I tried splitting the data into 2 different URL's the first time it seems to be executing properly but for the next function where I have the get 2nd URL it always gives out "Failed to complete HTTP get"

I'm using the ESP32_LTE_Demo as the baseline with modifications. Where is I just have to perform certain functions in order so got rid of the switch case() and added only the functions I wanted to carry out.
Whereas if I just use the ESP32_LTE_Demo and perform 'G' and '2' (multiple time) then it posts the data continuously without any issue.

Any inputs would be helpful.

[image]

Thank you.

bihmedical, 2019-06-13:
Ok I tried the building the code from scratch again and it seems to be working, I can send the data in chunks. I don't know what was exactly the problem with the older code.
Thank you for your responses.

closed 2019-06-13 by bihmedical