botletics/SIM7000-LTE-Shield #100: Cold Start / Warm Start / Hot Start

ttorkar, 2019-05-26:
Hi Tim,

I was wondering if you had any further information about Cold Start / Warm Start / Hot Start and how they operate? 
Is it simply a matter of `AT+CGNSHOT` or is there something more? Is everything handled by the SIM7000? 

Or does it require the XTRA file as per.

> SIM7000 Series_GNSS_Application
Note_V1.01

> It provides enhanced standalone performance, and eliminates the need to demodulate the GPS
> signal for ephemeris, almanac, iono, UTC, or health.
> Normally requires -144 dBm or stronger for all SVs in view.
> TTF can be reduced by 18 to 30 sec (or more in harsh signal environments) 

Thanks in Advance

botletics, 2019-05-26:
Sorry, besides what in the app notes I'm really not sure.

ttorkar, 2019-05-26:
Ok thanks, I'll let you know if I find anything out

closed 2019-05-26 by ttorkar

robbiet480, 2020-03-21:
Sorry to bump an old issue, but just wanted to point out that there are example commands in the GNSS app note to enable XTRA.

```
AT+SAPBR=3,1,"APN","hologram"
AT+SAPBR=1,1
AT+CNTPCID=1
AT+CNTP="pool.ntp.org",-28,1,2
AT+CNTP

AT+CNACT=1,"hologram"
AT+HTTPTOFS="http://xtrapath1.izatcloud.net/xtra3grc.bin","/customer/xtra3grc.bin"
AT+CGNSCPY
AT+CGNSXTRA=1
AT+CGNSCOLD
```
However, even after following the instructions, I get a URC `+CGNSXTRA: 2` which according to the AT commands manual means `XTRA file is not effective`. Unsure why.

robbiet480, 2020-03-31:
Turns out, you need to have a very good GPS signal for the XTRA file to work. Once I went outside and had clear view of the entire sky it instantly worked.