botletics/SIM7000-LTE-Shield #102: force LTE NB-IoT, needs enableGPRS?

tomtobback, 2019-06-03:
Hi, i live in Hong Kong and i'm trying to make my SIM7000C work with LTE and your modified FONA library. It defaults to 2G and GPRS works fine (so far tried 2 operators, CSL and China Mobile). These SIM cards also support 4G, and operators claim they have NB-IoT deployed, so i'm trying to force the SIM7000C to use LTE with setPreferredMode(38). I use setNetworkSettings with the APN, but do i still need to enable GPRS if i want to use LTE? When i do that, my module hangs after 'AT+CGATT=1'.
In the library .cpp file at enableGPRS() there are some comments 'uncomment for LTE only' but it seems they don't apply for the SIM7000.

botletics, 2019-06-03:
The lines that say "uncomment for LTE only" are applicable to the SIM7000 so you could try doing that. However, please first check if you are even connected to a network to begin with using the "n" and "1" commands in the LTE_Demo sketch

cejpmart, 2019-06-08:
I had to uncommend the LTE-only part and force mode 38 to get any GPRS at all (Lebara, the Netherlands). Otherwise I was getting stuck at either AT+CGATT or AT+CSTT

And at one point it also turned out that my battery wasn't supplying enough power, so be sure to check that out.

closed 2019-08-01 by botletics