botletics/SIM7000-LTE-Shield #103: SMS CMS Error 500

tonygomez, 2019-06-07:
using a SIM7000A, Hologram dashboard says device is connected usage is 0. I am in Raleigh, NC.

AT+CGREG?
Registered roaming

AT+CPSI?
LTE CAT-M1,Online,310-410,0x2717,195163152,49,EUTRAN-BAND12,5110,3,3,-10,-102,-76,14

LTE ONLY
AT+CNMP=38

CAT-M ONLY
AT+CMNB=1

AT+CMGF=1
AT+CMGS="19998889999"
> TEST <ctrl+z>
+CMS ERROR: 500

I expected this last command to send my mobile phone a sms message.  Thoughts on why the SMS is not sending?

tonygomez, 2019-06-07:
Latest findings

There are conflicting reports Network Registration command says Registration denied and the Network Registration Status command says Registered, roaming.  The hologram dashboard says connected.  Does registration denied explain why I cannot send an SMS?  How do you debug registration denied?

AT+CGREG?
+CGREG: 0,5
OK

AT+CREG?
+CREG: 1,3
OK

wmwilkins, 2019-06-07:
Tony,

Do you pay Hologram to get a phone number attached to the SIM?  If not, you
must use the soft SMS procedure.  There are instructions on the Hologram
website.

-Monty

tonygomez, 2019-06-07:
@wmwilkins I purchased a number within the last 15 minutes.  I have not seen anything regarding soft sms instructions. searching now for instructions....

wmwilkins, 2019-06-07:
It took me a little time to get it working and that was several months ago.  I  am not near my system but from what can recall soft sms uses Hologram's API & their cloud router to send the message.   If you have a phone  number I would think you could use the command line to send.

tonygomez, 2019-06-07:
@wmwilkins Thank you for the info. I will give it a bit and try more after purchasing number. Really appreciate your reply. The hologram forum seems very unresponsive - judging by others posts as well.

wmwilkins, 2019-06-07:
@tonygomez.  I think there could be some mention in this forum that some folks had issue with one of the network carriers (ATT or Verizon)  not allowing accepting SMS from roaming LTE-M endpoints but it worked when they used the command line to use the other carrier.

botletics, 2019-06-10:
Sorry for the late reply, but try forcing your SIM7000 to only use AT&T. You can use the command setOperatingBand("CAT-M",12) to force it to AT&T like in line 219 of the LTE_Demo code.

tonygomez, 2019-06-13:
I will give this a try soon. For the moment I abandoned SMS and using data.  (fyi, according to network status I was connected to AT&T)