botletics/SIM7000-LTE-Shield #105: Sending data post 3g to SIM5320A

RogerioTostes, 2019-06-19:
Hello,
I need help to send data via 3g http through the SIM5320A, I used the example LTE_DEMO, I made the changes to use the SHIELD 5320A, but I can not run the example of post 3g I am having the following error:

PB DONE
G
	---> AT+CGATT=1
	<--- OK
	---> AT+CGSOCKCONT=1,"IP","hologram"
	<--- OK
	---> AT+CIPMODE=1
	<--- OK
	---> AT+NETOPEN=,,1
	<--- Network opened
FONA> 3
	---> AT+CHTTPSSTART
	<--- OK
Waiting 1s to ensure connection...
	---> AT+CHTTPSOPSE="www.dweet.io",80,1
	<--- OK
Waiting 1s to make sure it works...
	---> AT+CHTTPSSEND=103
	<--- >
	---> GET /dweet/for/MSG_Texto1=66&MSG_Texto2=20&MSG_Texto3=10&MSG_Texto4=50 HTTP/1.1
Host: www.dweet.io

	<--- OK
	---> AT+CHTTPSSEND
	<--- OK
	<--- +CHTTPSSEND: 0
	---> AT+CHTTPSRECV?
	<--- +CHTTPSRECV: LEN,0
	---> AT+CHTTPSRECV=0
	<--- ERROR
	---> AT+CHTTPSCLSE
	<--- OK
	<--- 
	---> AT+CHTTPSSTOP
	<--- OK
Failed to complete HTTP/HTTPS request...

botletics, 2019-06-24:
One thing you can try is to increase the 1s delays to 2s and see if that helps. But it should just work.

RogerioTostes, 2019-06-24:
Thanks for replying, I did not find the code exactly where to change this delay, could you tell me where to change?

my code
[image]

botletics, 2019-06-24:
In the library .cpp file in the function "postData()"

RogerioTostes, 2019-06-25:
Hello again, following your suggestion I increased all delays to 5000 milliseconds and got this result:

[image]

Does this mean that the sent data has worked?

RogerioTostes, 2019-06-25:
Hello Timothy Woo, I was able to check and was successful in sending json. Thanks for the help. 
But you can help me in this other question I would like to use a rest api of firebase by sending a post like in the following example:
 curl -X POST -d '{"user_id": "jack", "text": "Ahoy!"} '\  'https: // [PROJECT_ID] .firebaseio.com / message_list.json'
is it possible to do that? Thank you again.

mainiiot, 2020-04-06:
Hello Timothy Woo, I was able to check and was successful in sending json. Thanks for the help.
But you can help me in this other question I would like to use a REST API of FireBase by sending a post like in the following example:
curl -X POST -d '{"user_id": "jack", "text": "Ahoy!"} '\  'https: // [PROJECT_ID] .firebaseio.com / message_list.json'
is it possible to do that? Thank you again.

????

botletics, 2020-04-06:
Yes, you just change the function parameters to do POST.

closed 2020-04-06 by botletics

mainiiot, 2020-04-06:
Thanks Timothy for the answer, one more question, how should I do in arduino to send httpparam to use firebase? because it tells me error 603.
And I can't find an example to stick with firebase using sim7000. Thanks you

  Serial2.println("AT+HTTPPARA=""URL"",""https://xxxxxxxxxxx.firebaseio.com/pages.json?auth=xxxxxxxxxxxxxxxxxxxxxxxxxx""");
  comdata = "";
  while ( comdata.indexOf("") < 0 )
  {
    delay(2);
    while (Serial2.available())
    {
      comdata = Serial2.readString();
      comdata.trim();
      Serial.println( comdata );
    }
  }
  Serial.println("");

botletics, 2020-04-06:
Firebase only accepts HTTPS so with SIM7000 you need SSL commands.

mainiiot, 2020-04-06:
Very true Timothy, I have not been clear and specific. 
Is there an example of how to use SSL Firebase with SIM7000? because the Simcom HTTPS document is definitely not clear and it has no examples, at least I have googled and haven't found, 
I am probably searching wrong, so I ask you in this forum.
Thank you very much Timothy

botletics, 2020-04-06:
There's no example specifically for Firebase but please see [this post] (https://github.com/botletics/SIM7000-LTE-Shield/issues/165)