botletics/SIM7000-LTE-Shield #107: Failing to read SMS messages

pilant, 2019-06-26:
I've been using this shield for a couple of weeks (mostly) without any problems.

Three times, to date, I have seen a state arise where the readSMS() method fails, even though there are messages to read (the getNumSMS() method returns a non-zero positive number).  Nothing seems to clear this state; removing the LiPo battery, Arduino Uno USB cable, etc.

So far, the only thing I have found to clear this state is to clear all SMS messages (AT+CMGD=0,4).

BTW, this has been verified using the LTE_Demo code as well as my own.

Any insights?

Thanks in advance.

\- Mark

botletics, 2019-07-23:
I haven't had this problem before. What firmware version do you have? Sounds like a firmware glitch or something.

pilant, 2019-07-23:
The results of an AT+CGMR are: Revision:1351B03SIM7000A

The SIMCom Library is V1.0.0, which I downloaded June 12, 2019; so it is pretty recent.

\- Mark

botletics, 2019-07-23:
Have you tried upgrading to 1351B04SIM7000A?

pilant, 2019-07-23:
I can give it a try.  Although, since I was not able to reliably reproduce the problem, I don't know of a clear way to know if the problem has been fixed.

botletics, 2019-07-23:
There's a possibility that it is a SIM card issue as well, where it glitches out somehow. Like you said, it's hard to say.