botletics/SIM7000-LTE-Shield #108: Sim7000 http post doesn't work if http get is not used before http post

bihmedical, 2019-06-27:
I have attached the code here. I am using ESP32 with Botletics sim7000 sheild and got this sketch from ESP32_LTE_Demo example. 
Http get works seemlessly buy when I use only http post method it seems to work intermittently. And when i do get before post then the code seems to be working fine. 
For my use purposes I have to make post method to work. Any help or suggestion would be useful. 
Thank you

`
         // Post data to website via 2G or LTE CAT-M/NB-IoT
        float temperature = analogRead(A0) * 1.23; // Change this to suit your needs

        uint16_t battLevel = 3600;

       
        char URL[150];
        char body[100];
        char tempBuff[16];
        char battLevelBuff[16];

        // Format the floating point numbers as needed
        dtostrf(temperature, 1, 2, tempBuff); // float_val, min_width, digits_after_decimal, char_buffer
        dtostrf(battLevel, 1, 0, battLevelBuff);

        // GET request
        sprintf(URL, "dweet.io/dweet/for/%s?temp=%s&batt=%s", imei, tempBuff, battLevelBuff); // No need to specify http:// or https://
        //        sprintf(URL, "http://dweet.io/dweet/for/%s?temp=%s&batt=%s", imei, tempBuff, battLevelBuff); // But this works too

       // if (!fona.postData("GET", URL))
       //  Serial.println(F("Failed to complete HTTP GET..."));

      
          sprintf(URL, "http://dweet.io/dweet/for/%s", imei);
          sprintf(body, "{\"temp\":%s,\"batt\":%s}", tempBuff, battLevelBuff);

        if (!fona.postData("GET", URL))
         Serial.println(F("Failed to complete HTTP GET..."));

        if (!fona.postData("POST", URL, body)) 
          Serial.println(F("Failed to complete HTTP POST..."));

            sprintf(URL, "http://dweet.io/dweet/for/%s", imei);
          sprintf(body, "{\"data1\":%s,\"data2\":%s}", tempBuff, battLevelBuff);

        if (!fona.postData("POST", URL, body)) 
          Serial.println(F("Failed to complete HTTP POST..."));`

botletics, 2019-07-01:
Please only use the lines relevant to the POST request and comment out the GET request code. It also looks like you are trying to post twice, one right after the other.

bihmedical, 2019-07-01:
I need to send multiple post requests as the data being sent is limited by 350 bytes. Is there any better way to post the data multiple times then?

botletics, 2019-07-23:
Sorry for the late reply. There is one thing you could do to increase the size:
- In the ESP32_LTE_Demo sketch, increase the size of the array that holds the POST data (it's called "body[]") and make it something like `body[450]` or however large you need.
- Next, depending on what baud rate you're using you may need to adjust the upload time in the file Adafruit_FONA.cpp line 1737 which has the line `sprintf(dataBuff, "AT+HTTPDATA=%d,10000", bodylen);` You can adjust the 10000 to something larger if needed, up to 120 seconds (120000ms).

bihmedical, 2019-07-26:
Thank you, I will give it a try and see if it works for our application.

JGLJGL, 2019-12-27:
Hi @bihmedical , did you achieve to do the POST ? If so, could you send the code ?

Thanks in advance,

JGLJGL, 2019-12-28:
Hi @botletics , could you help me with this problem? I need to get the POST request. Here it is part of the code 
```
void loop() {

 Serial.print(F("FONA> "));
 
  while (! Serial.available() ) {
    if (fona.available()) {
      Serial.write(fona.read());
    }
  }
          // Post data to website
   
        uint16_t statuscode;
        int16_t length;
        char url[150];
        char data[150];

        flushSerial();
        Serial.println(F("NOTE: in beta! Use simple websites to post!"));
        Serial.println(F("URL to post (e.g. httpbin.org/post):"));
        Serial.print(F("http://")); readline(url, 79);
        Serial.println(url);
        Serial.println(F("Data to post (e.g. \"foo\" or \"{\"simple\":\"json\"}\"):"));
        readline(data, 79);
        Serial.println(data);

        Serial.println(F("****"));
        if (!fona.HTTP_POST_start(url, F("text/plain"), (uint8_t *) data, strlen(data), &statuscode, (uint16_t *)&length)) {
          Serial.println("Failed!");

        }
        while (length > 0) {
          while (fona.available()) {
            char c = fona.read();

#if defined(__AVR_ATmega328P__) || defined(__AVR_ATmega168__)
            loop_until_bit_is_set(UCSR0A, UDRE0); // Wait until data register empty.
            UDR0 = c;
#else
            Serial.write(c);
#endif

            length--;
            if (! length) break;
          }
        }
        Serial.println(F("\n****"));
        fona.HTTP_POST_end();
        

      
// flush input
  flushSerial();
  while (fona.available()) {
    Serial.write(fona.read());
  }
}
```
and what gives the serial monitor:
```
12:10:28.794 -> NOTE: in beta! Use simple websites to post!
12:10:28.794 -> URL to post (e.g. httpbin.org/post):
12:10:28.794 -> http://www.atesens.com/post
12:10:40.286 -> Data to post (e.g. "foo" or "{"simple":"json"}"):
12:10:43.242 -> password
12:10:43.242 -> ****
12:10:43.276 -> 	---> AT+HTTPTERM
12:10:43.311 -> 	<--- OK
12:10:43.346 -> 	---> AT+HTTPINIT
12:10:43.381 -> 	<--- OK
12:10:43.416 -> 	---> AT+HTTPPARA="CID"
12:10:43.449 -> 	<--- OK
12:10:43.482 -> 	---> AT+HTTPPARA="UA"
12:10:43.515 -> 	<--- OK
12:10:43.582 -> 	---> AT+HTTPPARA="URL"
12:10:43.615 -> 	<--- OK
12:10:43.649 -> 	---> AT+HTTPPARA="REDIR"
12:10:43.682 -> 	<--- OK
12:10:43.751 -> 	---> AT+HTTPSSL=1
12:10:43.785 -> 	<--- ERROR
12:10:43.785 -> Failed!

```