botletics/SIM7000-LTE-Shield #109: ESP32+Botletics SIM7000 trying to issue 4 http post, it's working when USB is connected but FW goes in unknown loop when battery powered.

bihmedical, 2019-06-28:
Using ESP32+Botletics SIM7000. Started with ESP_LTE_Demo sketch everything seems to be working fine on the example. I created another code with the help of the example where I scan the BLE data convert it in Json format and send out the data in packets. Attached is the loop(). When I run this code and when the USB port is connected things seem to be working fine. As soon as I remove the USB connection, the ESP wakes up scans the BLE and SIM7000 board posts the data once and gets stuck is loop where it seems like it's waiting for a response? I'm not sure it's as if there a while loop and it never goes back to sleep again.
I know this based on current changes when board works fine it peaks 150-200mA during http post and sleep around 13mA but when the boards get stuck in a loop or something there's constant current draw of 70mA. There's no way for me to see the issues to debug if anything is going on because as soon as i connect USB things work fine. ESP sleeps for 1 min.
I would really appreciate some help. I tried adding delay after every http post it didn't work. 

`void loop() {
   BLE_default();
 
  Serial.print(F("FONA> "));
 
#ifndef SIMCOM_7000 
#endif

        if (!fona.enableGPRS(false))
          Serial.println(F("Failed to turn off"));
   
           // turn data off first for SIM7500
#ifdef SIMCOM_7500
        fona.enableGPRS(false);
#endif

        // turn data on
        if (!fona.enableGPRS(true))
          Serial.println(F("Failed to turn on"));
  
          
    // Serial.println(replybuffer);
#if defined(SIMCOM_2G) || defined(SIMCOM_7000)
 
         // Post data to website via 2G or LTE CAT-M/NB-IoT

        char URL[900];
        char body[700];
     
     //HTTP GET 
      // sprintf(URL, "dweet.io/dweet/for/%s?", imei);
     //  if (!fona.postData("GET", URL))
     //    Serial.println(F("Failed to complete HTTP GET..."));
      // delay(2000);
       
      std::string tempJson;
      
      //HTTP POST 1
      if(j>0 || j>1){
          // Serial.println(deviceid.c_str());
           location = replybuffer;
          // Serial.println(location.c_str());
           tempJson = "{\"id\":\"" + deviceid;
           tempJson += "\",\"loc\":[\"" + location;
           tempJson += "\"],\"data\":" + jsonData;
           tempJson += "}";
          // Serial.println("tempJson:");
          // Serial.println(tempJson.c_str());
      
          sprintf(URL, "http://dweet.io/dweet/for/%s", imei);
          sprintf(body, "%s", tempJson.c_str());
        
        if (!fona.postData("POST", URL, body)) 
          Serial.println(F("Failed to complete HTTP POST..."));
          delay(1000);
      }

     //HTTP POST 2
      if(j>2 || j>3 || j>4){
           tempJson = "{\"id\":\""  + deviceid;
           tempJson += "\",\"loc\":[\"" ;
           tempJson += "\"],\"data\":" + jsonData1;
           tempJson += "}";
         
           sprintf(URL, "http://dweet.io/dweet/for/%s", imei);
          sprintf(body, "%s", tempJson.c_str());

        if (!fona.postData("POST", URL, body)) 
          Serial.println(F("Failed to complete HTTP POST..."));
           delay(1000);
       }

     //HTTP POST 3
      if(j>5 || j>6 || j>7){
           tempJson = "{\"id\":\""  + deviceid;
           tempJson += "\",\"loc\":[\"" ;
           tempJson += "\"],\"data\":" + jsonData2;
           tempJson += "}";
         
           sprintf(URL, "http://dweet.io/dweet/for/%s", imei);
          sprintf(body, "%s", tempJson.c_str());

        if (!fona.postData("POST", URL, body)) 
          Serial.println(F("Failed to complete HTTP POST..."));
           delay(1000);
      }

     /*   //HTTP POST 4
       if(j>8 || j>9 || j>10){
           tempJson = "{\"id\":\""  + deviceid;
           tempJson += "\",\"loc\":[\"" ;
           tempJson += "\"],\"data\":" + jsonData3;
           tempJson += "}";
         
           sprintf(URL, "http://dweet.io/dweet/for/%s", imei);
          sprintf(body, "%s", tempJson.c_str());

        if (!fona.postData("POST", URL, body)) 
          Serial.println(F("Failed to complete HTTP POST..."));
       }             */
       
    #endif 
             

  flushSerial();
  while (fona.available()) {
    Serial.write(fona.read());
  }
   
     esp_sleep_enable_timer_wakeup(SLEEP_TIME * 1000000);    //new  (uint64_t)
     esp_deep_sleep_start(); 

}`

bihmedical, 2019-06-28:
I tried to change my code seems like GET method is working consistently but POST method is not. I hooked up both ESP and Botletics SIM7000 boards to separate power supplies. The ESP wakes up and sleeps on time but when ESP wakes up the current consumption on Botletics board is not changing as if it's not receiving any command over UART (this happens just during post method) and when the boards are not connected to battery. I really have to get this board working on POST method. Please provide some suggestions.

botletics, 2019-07-01:
Looks like you need to include code to shut down and wake up the SIM7000. I think it's hanging because you're all of a sudden closing the UART connection when the ESP32 goes to sleep and it doesn't know what to do when the ESP32 wakes up. So turn off or sleep the SIM7000 before sleeping the ESP32, then when the ESP32 wakes up, wake up the SIM7000 and do the UART initialization.

bihmedical, 2019-07-01:
fona.enableSleepMode(true);
D8 - This is the UART Data Terminal Ready (DTR) pin which can be used in conjunction with "AT+CSCLK" to wake the module out of sleep mode for low-power applications. For most users and general testing this pin may not be needed but it's there in case you do. To connect the pin, bridge the solder jumper with a little solder.
[image]

Using above 3 points I tried to put SIM7000 in sleep but it doesn't seem to be working. I connected D8 from sim7000 board to P19 on esp32. Issue low as soon as ESP wakes up on DTR , then fona.enableSleepMode(true); then issue low on DTR. After this I put esp32 to sleep so the DTR pin settings get erased and its driven low automatically. I have also shorted the jumper point provided on the sim7000 board to get DTR signal to pin D8. 
Please provide some inputs on this.
Thank you.

botletics, 2019-07-08:
Try not connecting DTR at all and try

`fona.enableSleepMode(true);
fona.set_eDRX(1, 4, "0010");`

JGLJGL, 2019-12-26:
Hi, I'm new and not able to make a POSTt request.
I'm working with the arduino IDE and attached to the ESP32 a SIM7000E.
I 've tried the "SIM7000-LTE-Shield-master"->"ESP32_LTE_Demo" , but not being able to make it work (POST request).

Have you been able to do the POST @bihmedical ? If so, could you send the code ?

Thanks in advance,

botletics, 2019-12-31:
The ESP32_LTE_Demo should work just fine. Please make sure you use the 'G' command first before using the '2' command to do a GET command. POST should also work.

JGLJGL, 2019-12-31:
Fails to turn on, so then, not able to complete neither the GET nor the POST request. Don't know if the issue could be related to Mobile Equipment failure or something related....

`12:46:15.319 -> FONA> n
12:46:19.931 -> 	---> AT+CGREG?
12:46:19.931 -> 	<--- +CGREG: 0,1
12:46:19.931 -> Network status 1: Registered (home)
12:46:19.931 -> FONA> i
12:46:28.963 -> 	---> AT+CSQ
12:46:28.963 -> 	<--- +CSQ: 13,99
12:46:28.963 -> RSSI = 13: -88 dBm
12:46:28.963 -> FONA> G
12:46:47.330 -> 	---> AT+CIPSHUT
12:46:47.377 -> 	<--- SHUT OK
12:46:47.424 -> 	---> AT+CGATT=1
12:46:47.424 -> 	<--- OK
12:46:47.471 -> 	---> AT+SAPBR=3,1,"CONTYPE","GPRS"
12:46:47.517 -> 	<--- OK
12:46:47.752 -> 	---> AT+SAPBR=3,1,"APN","orangeworld"
12:46:47.799 -> 	<--- OK
12:46:47.846 -> 	---> AT+CSTT="orangeworld","orange",""
12:46:47.892 -> 	<--- OK
12:46:47.939 -> 	---> AT+SAPBR=3,1,"USER","orange"
12:46:47.986 -> 	<--- OK
12:46:48.022 -> 	---> AT+SAPBR=3,1,"PWD",""
12:46:48.058 -> 	<--- OK
12:46:48.095 -> 	---> AT+SAPBR=1,1
12:46:48.130 -> 	<--- ERROR
12:46:48.130 -> Failed to turn on
12:46:48.130 -> FONA> 2
12:46:52.668 -> 	---> AT+HTTPTERM
12:46:52.715 -> 	<--- OK
12:46:52.715 -> 	---> AT+HTTPINIT
12:46:52.762 -> 	<--- OK
12:46:52.809 -> 	---> AT+HTTPPARA="CID",1
12:46:52.856 -> 	<--- OK
12:46:52.903 -> 	---> AT+HTTPPARA="URL","http://dweet.io/dweet/for/865234030907363?temp=28.70&batt=3600"
12:46:52.985 -> 	<--- OK
12:46:53.021 -> 	---> AT+HTTPACTION=0
12:46:53.058 -> 	<--- OK
12:47:03.026 -> Failed to complete HTTP GET...
12:47:03.097 -> 	---> AT+HTTPTERM
12:47:03.097 -> 	<--- ERROR
12:47:03.165 -> 	---> AT+HTTPINIT
12:47:03.165 -> 	<--- ERROR
12:47:03.165 -> Failed to complete HTTP GET...
12:47:03.202 -> 	---> AT+HTTPTERM
12:47:03.236 -> 	<--- ERROR
12:47:03.272 -> 	---> AT+HTTPINIT
12:47:03.319 -> 	<--- ERROR
12:47:03.319 -> Failed to complete HTTP POST...
12:47:03.319 -> FONA> ⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮
12:47:04.694 -> +HTTPACTION: 0,200,211
12:47:04.694 -> 
`

botletics, 2019-12-31:
Make sure you have a LiPo battery connected with the correct polarity. That is usually the problem for power issues. The red wire of the battery should be on the left side of connector when you plug it in.

JGLJGL, 2019-12-31:
I don't have exactly a LiPo battery, but I have a current supply to give voltage and use it as a battery.
Anyway, maybe I have something wrongly connected.
-ESP32 connected to the laptop through USB
-SIM7000 connected to the power supply through 5V and GND pins.
-ESP32 and SIM7000 connected to each other through Tx and Rx pins(SIM7000) with pins 16 and 17 (I have already changed them in the code)(ESP32) and GND pins.

botletics, 2019-12-31:
Like I said, a strong 5V supply on the "5V" pin won't power it adequately. You must have a strong 3.7V source connected to VBAT (please beware, 5V is too high on VBAT).

JGLJGL, 2019-12-31:
Ok thanks so much, I'will give it a try.

JGLJGL, 2020-01-03:
Hi again, I've tried to contact with the SIM7000 support service, but have not been able to.
I'm in doubt which the VBat pin is in this case. Is it possible that VBat is the same as PWR ? 

[Schematic:](https://www.waveshare.com/w/upload/6/61/SIM7000E-NB-IoT-HAT-Schematic.pdf)

This is what the board looks like:

[image]
[image]

botletics, 2020-01-03:
Sorry, you're not using the Botletics hardware.

JGLJGL, 2020-01-03:
Ok, but that doesn't mean I can't use the code I'm using with this hardware, does it?

botletics, 2020-01-03:
You're asking questions related to the hardware of the board itself, like pin definitions. I wouldn't know, sorry. Please contact Waveshare's customer support.

botletics, 2020-01-03:
(Also, as a side note, please note that the suggestions on power supply I mentioned earlier above will not necessarily apply to your case because I thought you were using a Botletics board.)