botletics/SIM7000-LTE-Shield #110: Library not compiling correctly on ESP32 with ESP-IDF and Arduino

leofabri, 2019-07-02:
- [x] ESP32 Wroom DevKitV1
- [x] Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
- [x] Compiler version: 5.2.0
- [x] Arduino is used as a module inside of IDF

Hi @botletics, 
I'm facing compilation issues with your library when I try to compile it for my ESP32 when using express-if and Arduino as a module. 
I get the following errors:

```cpp
CXX build/arduino/libraries/SIM7000-LTE-Shield//Adafruit_FONA.o
/esp//test/components/arduino/libraries/SIM7000-LTE-Shield/Adafruit_FONA.cpp: In member function 'boolean Adafruit_FONA::getGPS(float*, float*, float*, float*, float*, uint16_t*, uint8_t*, uint8_t*, uint8_t*, uint8_t*, float*)':
/esp//test/components/arduino/libraries/SIM7000-LTE-Shield/Adafruit_FONA.cpp:1054:11: warning: variable 'tok' set but not used [-Wunused-but-set-variable]
     char *tok;
           ^
/esp//test/components/arduino/libraries/SIM7000-LTE-Shield/Adafruit_FONA.cpp: In member function 'boolean Adafruit_FONA::postData(const char*, const char*, const char*, const char*, uint32_t)':
/esp//test/components/arduino/libraries/SIM7000-LTE-Shield/Adafruit_FONA.cpp:1696:23: error: comparison with string literal results in unspecified behaviour [-Werror=address]
   if (request_type == "GET") {
                       ^
/esp//test/components/arduino/libraries/SIM7000-LTE-Shield/Adafruit_FONA.cpp:1700:28: error: comparison with string literal results in unspecified behaviour [-Werror=address]
   else if (request_type == "POST" && bodylen > 0 ) { // POST with content body
                            ^
/esp//test/components/arduino/libraries/SIM7000-LTE-Shield/Adafruit_FONA.cpp:1727:28: error: comparison with string literal results in unspecified behaviour [-Werror=address]
   else if (request_type == "POST" && bodylen == 0) { // POST with query parameters
                            ^
/esp//test/components/arduino/libraries/SIM7000-LTE-Shield/Adafruit_FONA.cpp:1731:28: error: comparison with string literal results in unspecified behaviour [-Werror=address]
   else if (request_type == "HEAD") {
                            ^
/esp//test/components/arduino/libraries/SIM7000-LTE-Shield/Adafruit_FONA.cpp: In member function 'boolean Adafruit_FONA::postData(const char*, uint16_t, const char*, const char*, const char*)':
/esp//test/components/arduino/libraries/SIM7000-LTE-Shield/Adafruit_FONA.cpp:1763:13: error: suggest parentheses around comparison in operand of '|' [-Werror=parentheses]
   if (_type == SIM7500A | _type == SIM7500E) {
             ^
/esp//test/components/arduino/libraries/SIM7000-LTE-Shield/Adafruit_FONA.cpp:1797:13: error: suggest parentheses around comparison in operand of '|' [-Werror=parentheses]
   if (_type == SIM7500A | _type == SIM7500E) {
             ^
/esp//test/components/arduino/libraries/SIM7000-LTE-Shield/Adafruit_FONA.cpp:1828:13: error: suggest parentheses around comparison in operand of '|' [-Werror=parentheses]
   if (_type == SIM7500A | _type == SIM7500E) {
             ^
/esp//test/components/arduino/libraries/SIM7000-LTE-Shield/Adafruit_FONA.cpp: In member function 'boolean Adafruit_FONA::FTP_PUT(const char*, const char*, char*, size_t)':
/esp//test/components/arduino/libraries/SIM7000-LTE-Shield/Adafruit_FONA.cpp:2117:20: warning: comparison is always false due to limited range of data type [-Wtype-limits]
       if (remBytes >= 300000) {
                    ^
/esp//test/components/arduino/libraries/SIM7000-LTE-Shield/Adafruit_FONA.cpp: In member function 'boolean Adafruit_FONA::MQTTunsubscribe(const char*)':
/esp//test/components/arduino/libraries/SIM7000-LTE-Shield/Adafruit_FONA.cpp:2386:1: error: no return statement in function returning non-void [-Werror=return-type]
 }
 ^
/esp//test/components/arduino/libraries/SIM7000-LTE-Shield/Adafruit_FONA.cpp: In member function 'boolean Adafruit_FONA::MQTTreceive(const char*, const char*, int)':
/esp//test/components/arduino/libraries/SIM7000-LTE-Shield/Adafruit_FONA.cpp:2390:1: error: no return statement in function returning non-void [-Werror=return-type]
 }
 ^
/esp//test/components/arduino/libraries/SIM7000-LTE-Shield/Adafruit_FONA.cpp: In member function 'boolean Adafruit_FONA::MQTTdisconnect()':
/esp//test/components/arduino/libraries/SIM7000-LTE-Shield/Adafruit_FONA.cpp:2394:1: error: no return statement in function returning non-void [-Werror=return-type]
 }
 ^
At global scope:
cc1plus: warning: unrecognized command line option '-Wno-frame-address'
cc1plus: some warnings being treated as errors
make[1]: *** [/esp/esp-idf/make/component_wrapper.mk:290: libraries/SIM7000-LTE-Shield//Adafruit_FONA.o] Error 1
make: *** [/esp/esp-idf/make/project.mk:582: component-arduino-build] Error 2
```

It seems like the compiler is having issues with the way the comparisons are done (maybe the compiler is expecting strcmp?), but I don't understand why that is happening. I didn't test it with the bare Arduino IDE but I suppose that it works just fine. 
Thank you in advance for your support

leofabri, 2019-07-02:
Hello, I tried to apply some changes to Adafruit_FONA.cpp (see pull between requests) and the code compiles just fine on my ESP32. Unfortunately though, when I try to connect to the SIM7600 module I experience the following problem:
```
Initializing connections....(May take several seconds)
Configuring to 9600 baud
Attempting to open comm with ATs
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- OK
        ---> AT
        <--- 
        ---> AT
        <--- OK
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
Timeout: No response to AT... last ditch attempt.
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> AT
        <--- 
        ---> ATE0
        <--- 
        ---> ATE0
        <--- 
Couldn't find FONA
````

As you can notice, after some attempts, the module replies with "OK" to the "AT" request but seems to be completely ignored. So after some attempts I just get the "Couldn't find FONA" error. I don't know what's going wrong.
I tried looking inside the Adafruit_FONA.cpp and I noticed that you are performing a comparison using the `sendCheckReply(sendbuff, ok_reply)` function that seems to return **false** every time it gets called. 
Is it an expected behavior or am I looking at the wrong piece of code?
 
Please help. 
Thank you.

leofabri, 2019-07-02:
Okay, I think I'm starting to understand how things work. `sendCheckReply(sendbuff, ok_reply)` seems to work fine. It's something else, it's taking a lot of time to connect. 

#### This is the current workaround
I have done the following changes to the code that checks whether the SIM module is detected or not. Instead of "giving up" and report that FONA is not connected, I continue to iterate till something is detected (basically forever). 
```cpp
  if (! fona.begin(fonaSS)) {
    Serial.println(F("Couldn't find FONA"));
    // while (1); // Don't proceed if it couldn't find the device
    // Instead of getting stuck here, just try again and see if the module connects or not
    delay(100);
    if(fona.begin(fonaSS)) 
      return;
  }
```
After about 3 failed attempts, the module is detected successfully and appears to be recognized (I can read the IMEI of it). 
What I struggle to understand is why it takes this much time to connect... it's very strange. I tried to upload an untouched sketch from the Arduino IDE (so without expressif) and it immediately connects.

botletics, 2019-07-22:
Sorry, I've never really used ESP-IDF with Arduino. Are you using it so that you can include the IDF commands inside Arduino IDE? Are you saying that simply installing ESP-IDF makes the code not compile in Arduino IDE, without even adding any IDF commands inside?

eueduardo3, 2019-09-14:
> Hello, I tried to apply some changes to Adafruit_FONA.cpp to make it runnable (see pull between requests) and now it seems to compile just fine on my ESP32. Unfortunately, when I try to connect to the SIM7600/SIM7000G modules I experience the following problem:
> 
> ```
> Initializing connections....(May take several seconds)
> Configuring to 9600 baud
> Attempting to open comm with ATs
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
>         ---> AT
>         <--- OK
>         ---> AT
>         <--- 
>         ---> AT
>         <--- OK
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
> Timeout: No response to AT... last ditch attempt.
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
>         ---> AT
>         <--- 
>         ---> ATE0
>         <--- 
>         ---> ATE0
>         <--- 
> Couldn't find FONA
> ```
> 
> Something else was written here. Though I decided to clean my help request a little bit after playing around with the library for several hours without any result.
> 
> The only thing I can say about this is that if you upload the code using just the Arduino IDE without esp-idf everything works as it should. The sim module gets detected almost immediately and nothing bad seems to happen. However when you use esp-idf nothing works. The result is the one above.
> I cannot get the sim module connected because it's not seen by the board.
> 
> Please help.
> Thank you.

Hello, could you please make available (share) the changes you made to work with your ESP32 (using SIM700 or whatever, I have the other models in the SIM7xxx line)?

ishdemon, 2021-08-28:
@leofabri I also imported this repo on platformio, compiled fine for esp32 but having same problem as you.. AT Test fails and it can't detect the module..