botletics/SIM7000-LTE-Shield #112: SIM7000A stock firmware does not support HTTPS?

iamchriskelley, 2019-07-12:
Hi,

I recently ordered a SIM7000A shield from Amazon. This shield is a "version 5" design and shipped with "B01" firmware. 

None of the commands in the [HTTPS AT commands guide](https://github.com/botletics/SIM7000-LTE-Shield/blob/master/SIM7000%20Documentation/Technical%20Documents/SIM7000%20Series_HTTPS_Application%20Note_V1.00.pdf) are recognized by the modem.

To confirm this issue: send any of these commands to the modem in test mode. For example, the command "AT+SHCONF=?" should yield a reply of "+SHCONF: <HTTPParamTag>,<HTTPParamValue>\n\nOK" even if the modem is not ready for HTTPS commands. My modem however issues "ERROR" to all 12 HTTPS commands in the documentation.

Please confirm whether a SIM7000A shield purchased from you on Amazon can actually handle HTTPS without a firmware upgrade.

Thank you

botletics, 2019-07-12:
Correct, HTTPS won't work on that. The v6 version comes with 1351B03 for the SIM7000A, 1529B01 for the SIM7000G and 1351B05 for the SIM7000E. The SIM7000G one on Amazon is a guaranteed v6 version but the others could be v5 or v6 (although more likely v6).

iamchriskelley, 2019-07-12:
Okay, thanks for the clarification. Is there anywhere to definitely purchase a v6 SIM7000A shield? I went through the firmware update process yesterday and the update tool got stuck first at the IMEI verification stage, and then later at the firmware download stage (much like the description of [issue #41](https://github.com/botletics/SIM7000-LTE-Shield/issues/41)) and I may now have bricked my modem.

Also, given the description is [issue #66](https://github.com/botletics/SIM7000-LTE-Shield/issues/66), could you please state which firmware version(s) for the SIM7000A are proven to support the HTTPS/SSL AT command set?

botletics, 2019-07-12:
So you're using the SIM7000E? If you get the [SIM7000G shield on Amazon](https://www.amazon.com/dp/B07H3YNCQS/ref=cm_sw_r_cp_awdb_t1_qAikDbXXPCWV8) will for sure be version v6. I'm verifying with SIMCom which firmware versions support HTTPS.

iamchriskelley, 2019-07-12:
No, I'm using the SIM7000A, but good point about the SIM7000G -- that has the 1529B01 firmware? Thank you for checking with SIM; please let me know if 1529B01 firmware definitely supports the HTTPS command set

botletics, 2019-07-12:
Yep, it has 1529B01 firmware. I'll check with them and get back to you.

iamchriskelley, 2019-07-17:
Any word from SIMCom? Thanks

botletics, 2019-07-18:
Sorry, not yet. I'll bug them again.

botletics, 2019-07-22:
So no word from SIMCom yet but I (partially) tested the SIM7000G and SSL commands seem to be supported on the v6 boards from Amazon with 1529B01SIM7000G firmware.

iamchriskelley, 2019-08-05:
So I got eager and purchased a SIM7000G board. The v6 board's firmware does indeed support the SSL commands suite, however I couldn't make a simple GET command work, and as other people have noted the URL length limit is stated as 64 characters, and for a POST request while the SIMCom documentation lists the fields of the AT+SHCONF and AT+SHBOD commands at 350 characters I found that with the 1529B01SIM7000G firmware the fields are actually limited to 64 characters. So, provided you have the certificates worked out in advance, you should be able to use the demo HTTPS requests outlined in their application note, but the commands are of pretty limited real-world utility.

It looks from the different versions of the SIM7000 HTTPS AT Commands note that the field length for key config commands has increased over time, so a firmware upgrade might help solve what is effectively a major bug. I've switched to TCP requests for my current project though, so I haven't tested this.

ericyoung730, 2020-02-11:
I recently ordered a SIM7000C shield. I want to ask whether the SIM7000C supports the HTTPS function? Here are my testing results.

AT+GMR                                                                          
Revision:1351B08SIM7000C                                                        
                                                                                
OK   
AT+CNACT?                                                                       
+CNACT: 1,"10.107.154.239"                                                      
                                                                                
OK  
AT+CSSLCFG="convert",2,"foo.crt"                                           
OK 
AT+SHSSL=1,"foo.crt"                                                       
OK
AT+SHCONF?                                                                      
+SHCONF:                                                                        
URL: https://foo.com:443                                                    
TIMEOUT: 60                                                                     
BODYLEN: 350                                                                    
HEADERLEN: 350                                                                  
POLLCNT: 15                                                                     
POLLINTMS: 500                                                                  
IPVER: 0                                                                        
                                                                                
OK   

AT+SHCONN                                                                       
ERROR 

Thank you

Francois-37, 2021-05-06:
Hello everybody,

I know that subject is old but I have also some questions about the firmware witch support the SSL. I must admit that I am a little lost in all the firmware versions...

I have a SIM7000G, Revision:1529B04SIM7000G that work in SSL (CACID work but not the HTTPS AT commands guide).
I have also a SIM7000E, Revision:1351B03SIM7000E that doen't work in SSL (neither CACID, nor the HTTPS AT commands guide).

My question is, which version of the firmawre for the SIM7000E must I install in order that the SSL work, specificly the number 18 of the datasheet of the SIM7000, AT Commands for TCP/UDP Application Supported SSL.

Thank you for helping.

TimoWielink, 2021-07-05:
Hi everyone, I have a SIM7000G and trying to use HTTPS, Postdata works fine with HTTP but switching to HTTPS wont work.

Are there any flags or methods I need to set for HTTPS for the SIM7000G?

```
 char URL[200];
      char body[256];
      sprintf(URL, "TEST URL");
      sprintf(body, jsonstat.c_str());
      if (!fona.postData("POST",443, "HTTPS", URL, body)) // Can also add authorization token parameter!
      Serial.println(F("Failed to complete HTTP POST..."));
```

botletics, 2022-09-08:
Hey guys, figured out how to connect with SSL without verifying certs! Please do the following:
- Set "SSL_FONA" to 1 in the .h file
- Download the latest .cpp file from GitHub
- Open the unedited LTE_Demo example sketch and change "http://dweet.io" to "https://dweet.io" on line 1035 (under the '2' option)
- Upload to the Arduino
- Run the 'G' command to enable data connection, then run '2' to send data to dweet.io using SSL. And voila! 😊