botletics/SIM7000-LTE-Shield #115: B02 Firmware for SIM7000G

chaton78, 2019-07-24:
Hi,

I have some issues connecting to Rogers in Canada with Hologram SIM. I was able once... I am wondering if you can grab the latest firmware B02 for the SIM7000G. Perhaps there is something resolving my issue.

https://techship.com/downloads/category/firmware-firmware-simcom-simcom-sim7000g/

Thank you,

botletics, 2019-07-24:
Have you tried setting setPreferredMode(38) to use CAT-M1 only, then maybe setOperatingBand("CAT-M",12) except with the right band number for Rogers in Canada?

Last I heard (which was a while ago) the Hologram SIM card does not support LTE CAT-M networks in Canada so you might have to get a SIM card from Rogers.

chaton78, 2019-07-31:
setPreferredMode did the trick... I've not set the Operating Band.

Testing the stability... something to note: I am not connected to the closest likle when I connect with another GPRS device. I am connected to a tower much farther. 

Thank you!

closed 2019-07-31 by chaton78

botletics, 2019-07-31:
LTE CAT-M is not necessarily the same as 2G. I would imagine the tower you're connected to would be the closest one that supports CAT-M.