botletics/SIM7000-LTE-Shield #116: SIM7600 support

Vicky-S, 2019-07-31:
HI,

Does the libraries will work for SIM7600 module?

botletics, 2019-07-31:
No, it does not currently support the SIM7600, although it may be easily modified to do that since it probably shares many of the same commands.

Vicky-S, 2019-08-01:
ok. thanks. The good thing about SIM7600 is it, supports major 4g bands so we can use any SIM card for the operation. It will be very much helpful if you can merge this one.

botletics, 2019-08-01:
I just updated the library files. Please let me know if it works for you. If you could post the serial monitor results that would be greatly appreciated. The features that should work are recognizing that it's a SIM7600 module and communicating with it, all the basic stuff like network registration, network info, etc. (go through the menu in LTE_Demo.ino sketch), and posting data via 4G.

closed 2019-08-01 by botletics

botletics, 2019-08-01:
Voice calling should work as well.

hunterirving, 2019-08-20:
Hi botletics, Vicky-S,

When you say voice calling should be working, is that in reference to the SIM7000-LTE-Shield or another configuration involving the 7600?

Thanks in advance!
Hunter

botletics, 2019-08-26:
That's referring to the SIM7500/7600.

SL4P, 2020-03-18:
I just received some 7600 modules, and all my existing code works perfectly - EXCEPT for incoming and outbound VOICE commands/notifications..?
Any hints on what’s changed ?  or that I need to change.
Is there something I need to setup, or a pin that needs to be biased before using voice ?
ATD ;  ATH commands. RING +CLIP notifications etc just don’t happen.