botletics/SIM7000-LTE-Shield #117: Cant use MQTTs

khiemnt00, 2019-08-02:
Hi all,
I am using SIM700E version B08. When using SSL, I have 2 issues:
First, I only want to use the CA cert file, after AT+CSSLCFG convert, I used SMSSL command to select the CA cert file, but it always returns ERROR ( log below):
AT+CSSLCFG=convert,2,cacert.pem
OK

AT+SMSSL=0,cacert.pem 
ERROR

AT+SMSSL?
+SMSSL: 0,"","" 
OK 

I have tried many times but I still can't find what wrong with that, so I changed the method, using SSL with CA cert file, Client cert file, and key file. Everything seems to be ok, but when I using SMCONN command to connect to the server, it returns ERROR:
AT+CSSLCFG=convert,2,cacert.pem
OK

AT+CSSLCFG=convert,1,cert.pem,keyrsa.key
OK 

AT+SMSSL=1,cacert.pem,cert.pem
OK

AT+SMSSL?
+SMSSL: 1,"cacert.pem","cert.pem" 
OK 

AT+SMCONN
ERROR
When I use 3 files above with MQTT.fx, it works, I can connect to the server. 
So what's wrong with those commands? And how to fix it?
Thanks ðŸ™‚

botletics, 2019-08-02:
Hi, please only [post in one place](https://community.botletics.com/post/cant-use-mqtts-10205022?trail=15#1).

closed 2019-08-02 by botletics

zibyan123, 2022-09-22:
I am  trying to connect sim7000g with aws but I am not able to convert certificates .AT+CSSLCFG="convert"2,"root.pem" return ERROR.Is there any solution?

botletics, 2022-12-04:
For HTTPS, please [see this info](https://github.com/botletics/SIM7000-LTE-Shield/issues/156#issuecomment-1335989351)