botletics/SIM7000-LTE-Shield #122: UART connection on RPi 3B+

TTimTT, 2019-08-07:
Hello,

First of all, thank you very much for all this work, it is really helping!
I have unfortunately an issue when I try to use the Shield with **Raspbian on RPI 3B+** using the UART port only.

Everything works fine, I can use AT command, even Modem-Manager helps me to connect to my APN, but then I can not create an interface with Network-Manager, it timeouts.

By the way, everything works fine when using the USB, it looks like it is related to the UART?

I have read some previous issues and I know you are not a *"Linux guy"*, so I was hoping you could redirect me to someone who might know.

Thank you very much for your help!

bliteknight, 2019-08-07:
is UART your only option? 
You can get a network connection via USB which in my case is more reliable and gives you more options to send  data.

TTimTT, 2019-08-07:
Yes USB is great I agree. Unfortunately I must use UART, we designed our PCB that way.

wille-io, 2020-12-28:
@TTimTT Did you manage to get it working? If so, how?
Thank you.

TTimTT, 2020-12-28:
@wille-io  Hey, well in the end we used the USB connection, we never managed to make the UART  work properly, I would highly recommend also using the USB. 

In our case, we fix this by using 2 wires and connect them from the RPi USB pad directly to the SIM700-LTE USB pads.

Hope it helps,