botletics/SIM7000-LTE-Shield #123: Couldn't find FONA

Virginia2006, 2019-08-07:
Hi,

The following message "FONA could not be found" is displayed when I run LTE-demo, I bought the SHIELD directly in Botletics. The SIM I am using is from another operator (I cannot use the Hologram SIM) and I have already configured the APN of this operator. Can you help me solve this problem?

botletics, 2019-08-07:
Please make sure you plug in a 3.7V LiPo battery into the shield and check the connections on pins 10 and 11.

Virginia2006, 2019-08-07:
After compiling the following message appears,  what
What do I need to verify from pins 10 and 11? I soldered the arduino to the shield
[image]

botletics, 2019-08-07:
Yes, looks like it compiled fine. I just meant to say that if it's not finding the module, maybe check the solder on pins 10 and 11 and make sure you have power from a battery.

botletics, 2019-08-07:
Also, please consider posting on the [Botletics forum] (https://community.botletics.com/) instead.

Virginia2006, 2019-08-07:
Is it necessary to plug in 3.7V LiPo battery into the shield?

botletics, 2019-08-08:
Yes, definitely.

closed 2019-08-08 by botletics