botletics/SIM7000-LTE-Shield #124: SIM7000G with T-Mobile 2G

cszuo, 2019-08-14:
I got a [SIM7000G](https://www.amazon.com/gp/product/B07H3YNCQS/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) from amazon. The reason I purchase SIM7000G instead of SIM7000A is that it can work for both IoT network(NB-IoT,CAT-M) and 2G. It works great with hologram IoT SIM card. But it doesn't work with my T-Mobile SIM card from my phone. First of all, I have tested that my SIM card can work under GSM(2G) with module [A9G](https://www.amazon.com/Development-Quad-band-Wireless-Transmission-Geekstory/dp/B07BQZ8T9B/ref=sr_1_3?keywords=A9G&qid=1565751682&s=electronics&sr=1-3) there is not problem to send data or SMS. So this SIM7000**G** should work with GSM card, I have tried AT command like **AT+CNMP=13** which will case the module keep rebooting after few seconds. Also tried to set APN like what I did for the A9G module, but still doesn't work. 

So, does SIM7000G support 2G in USA, it looks like [they](https://github.com/botletics/SIM7000-LTE-Shield/issues/102) can connect to 2G. Or did I missed any AT commands? Thank you.

botletics, 2019-08-19:
I think 2G is no longer supported by T-Mobile. The main reason of using the SIM7000 in the US is to utilize LTE CAT-M or NB-IoT

closed 2019-08-19 by botletics

cszuo, 2019-08-19:
I am pretty sure 2G is working in my area by T-Mobile as I have successfully used it with GPRS module A9G.

Another problem is I am not able to use SIM7000G to receive SMS from Hologram, but I can receive SMS with SIM7000A. Do you have any idea?

botletics, 2019-08-19:
Hmmm I'm not sure about the 2G. I think they're at least discontinuing it on 2020.

If you can't connect to the network you won't receive SMS, which makes sense. But if you were connected, perhaps on the SIM7000G you were connected to Verizon. Try using setBandPreference() with band 12 to force it to AT&T. Some users have reported that on Verizon SMS aren't received properly.