botletics/SIM7000-LTE-Shield #126: FOTA firmware delta file

mplebanski, 2019-08-30:
Do you have delta files mentioned in [FOTA document](https://simcom.ee/documents/SIM7000x/SIM7000%20Series_FOTA_Application%20Note_V1.00.pdf)? I would like to incrementally update my module over the air starting from 1351B05SIM7000E and ending at 1351B08SIM7000E. Have you guys tested this procedure before?

madatr, 2019-11-18:
Hello mplebanski,

Did you attempt the FOTA procedure? if you did would you please tell me how did it go, I get 601 result from AT+HTTPTOFS command stated in the manual, I tried multiple version numbers in place of V1__V2 in the url but to no end.

It seems that the ip is functional as a ping is returned when attempted to it, the issue might be in the URL or my setup of the connection.

Please help,

Thanks.

botletics, 2019-11-18:
Have you looked at [the firmware files on GitHub](https://github.com/botletics/SIM7000-LTE-Shield/tree/master/SIM7000%20Documentation/Firmware)?