botletics/SIM7000-LTE-Shield #127: uploading code to Ardunio uno R3 Too much memory is used

dudemyass, 2019-08-31:
When i upload the example code IoT_Example | Ardunio 1.8.9 i get a message that 77% of memory has been used and it may cause stability problems. When i open the serial window it finds the fona module registers with the network and stops at that point and does not execute the rest of the code? Nothing has been changed in your code except the apn. When tested with other example LTE it posts to dweet.io if i select option 2 and all other options seem to work correctly only the IoT example is not working. I am using Botletics sim7000A ver 6 and i am in the USA.

botletics, 2019-09-02:
A sure-fire fix is to switch to a Mega, but if that isn't an option you could try minimizing the code to only the stuff you need.

alberg79, 2020-01-19:
I have the same issue. The Iot_Example is hanging in the moduleSetup function and I did try a Mega along with a Uno.  I added some serial print messages as flags to debug and found the hangup is at the get Imei number section of moduleSetup.  I commented it out and all I ended up with was a looping reboot part way through serial printing "Found Sim7000A " so the problem moved higher in the code.  LTE_demo sketch works great as does the MCP9808 example.  I'm learning as I go and will update if I can figure it out.

tjdzor, 2020-01-23:
I too had been fighting this issue, but figured a couple things out!

Background, trying to make a cellular temperature logger using AdafruitIO for a vacant property that has had some random heating issues.  Using a Hologram sim, everything worked fine in the SIM7XXX_LTE_Demo, but would go into a reboot loop or hang if I tried using the AdafruitIO_MQTT_Demo or IoT_Example.  Tried the previously mentioned solutions such as adding a battery, bought a Mega to try thinking maybe it was a stability issue.  But alas, no love.

So I added some serial outputs and tried commenting sections out to help figure out where it was failing.  I found two issues in `setup()`

It seems to hang @ `tempsensor.wake();`
I looked at the [Adafruit MCP9808](https://github.com/adafruit/Adafruit_MCP9808_Library/blob/master/examples/mcp9808test/mcp9808test.ino) example and sure enough, theres no wake before initializing it.  So I commented out that line. (`//tempsensor.wake();`)

And next, it would fail to find the MCP9808 anyways.  It's being started without the i2c address, so adding the address 0x18 fixed it.  The address can be checked with an [i2c Scanner](https://playground.arduino.cc/Main/I2cScanner/).  And the sensor can be tested with the [Adafruit MCP9808 Library/Example](https://github.com/adafruit/Adafruit_MCP9808_Library)
`if (!tempsensor.begin(0x18)) {`

I also set the resolution of the temp sensor to the highest (but also slowest), but that probably isn't necessary.
`tempsensor.setResolution(3);`

Mine has been working flawlessly for a couple days now.  Would love to try it on an Uno, but due time constraints its staying on the Mega for the time being.

I really hope this helps some people!

alberg79, 2020-01-25:
Uno too much memory message has nothing to do with the hang in the IoT_Example sketch.  As mentioned above the fix is to comment out the tempsensor.wake(); line.