botletics/SIM7000-LTE-Shield #130: SIM7000C stuck in "Factory Test Mode"

logaeshKTBS, 2019-09-23:
My sim7000C module works fine till today morning. After creating some mess with at commands my module stucks in FTM(Factory Test Mode). Whenever i issued AT+CFUN? command via uart i got "+CFUN: 5" as a response. I can't issue "AT+CFUN=1" since it throws error. 
I have issued "AT&F0" (Factory reset), but nothing helps. Please help me to get out from FTM.

botletics, 2019-09-23:
Worst case, try performing a firmware update and abort it after it resets the module.

logaeshKTBS, 2019-09-23:
Thank you for your reply. Still getting "+CFUN: 5" after restarting using "AT+CFUN=6". 
I will try to update the frimware, but i had some queries,

1. We have designed a custom board with sim7000C and a 32-bit arm controller. Unfortunately we didn't take the USB pins(28, 27) out of the board. I only have access to UART pins(9, 10). Is it possible to upgrade the frimware using UART?

2. I didn't find any fimware file for sim7000C in your GitHub (SIM7000-LTE-Shield/SIM7000Documentation/Firmware/) page. where can i find one for SIM7000C?

botletics, 2019-09-23:
Sorry, I am not sure if you can update firmware via UART. You may be able to on UART3.
You may want to get the file from SIMCom. I do not have SIM7000C firmware.

closed 2019-09-23 by botletics

hasikp1, 2019-11-22:
@botletics I got the same issue after a firmware update in SIM7000C. The update was successful but the module stuck at FTM

hasikp1, 2019-11-22:
Got it resolved, you have to restore xQCN file using QPST tool

eveready1010, 2020-01-15:
I have the same problem...  Upgraded the firmware (I think) and now it is stuck in factory test mode.  I tried all of the above and a few other suggestions at other sites, but no luck.  I have a battery pack installed.  I do not have a xQCN file from prior to the backup.
When I tried any of the other modes, the command looks like it succeeds, but after reboot it is still in mode 5.
Not sure what else to do.

eveready1010, 2020-01-18:
Anyone?  Not sure what else to do.  Google searches only turn up a few pages and none have worked so far.

botletics, 2020-01-18:
I think all you have to do is run the updater tool again and once it gets to a certain point just about it. You can also try the QPST tool.

eveready1010, 2020-02-22:
Ok, I have gotten the SIM7000 to go into CFUN=1, but now it will not connect:

 

Through Serial interface:

+CFUN: 1

+CPIN: READY

+CREG: 0,0

+CSQ: 31,99

+CPSI: NO SERVICE,Online

 

Through Arduino:

SIM7000/SIM7500 LTE Demo

Configuring to 9600 baud

Attempting to open comm with ATs

              ---> AT

              <--- 

              ---> AT

              <--- AT

              ---> ATE0

              <--- ATE0

              ---> ATE0

              <--- OK

              ---> ATI

              <--- SIM7000A R1351

 

OK

 

              ---> AT+CPMS="SM","SM","SM"

              <--- +CPMS: 0,10,0,10,0,10

FONA is OK

Found SIM7000A (American)

              ---> AT+GSN

              <--- 8652350xxxxxxxx

Module IMEI: 8652350xxxxxxxx

              ---> AT+CFUN=1

              <--- OK

              ---> AT+CGDCONT=1,"IP","hologram"

              <--- OK

FONA> G

              ---> AT+CIPSHUT

              <--- SHUT OK

              ---> AT+CGATT=1

              <--- 

Failed to turn on

FONA> 

ERROR

I have also followed several of your “Getting Started” and other documents online several times and different ways and yet the result is still the same.

So, I am still wondering if although I got the firmware upgraded and working again maybe something still went wrong???

Also, in the Hologram.io website, it shows the device as active, but not connected and no IMEI number is shown after all of the extensive testing I performed.

 

At this point I am thinking that the module itself is corrupt…  Or, could it be something as simple as a bad antennae? Signal strength seems ok though.

 

Any suggestions?

 

Thank you.

 

Ed Thomas

botletics, 2020-02-24:
Check the "Connect to network" wiki page on GitHub. You may need to send the AT+CGDCONT command as well as AT+CMNP or AT+COPS to get connected.

eveready1010, 2020-02-24:
Well, that helped greatly!!!

 

It seems that somehow the module was stuck in manual mode (mode 1) for the command: AT+COPS

 

Once I restarted the module and then set it to mode 0, magically it started to work!

 

You don’t know how frustrating it was not to be able to get it to work after following all of those wiki articles several times before.

 

Thank you for your help!!!

 

(I guess persistence pays off!)

 

Ed

wicadmin, 2020-07-09:
> Ok, I have gotten the SIM7000 to go into CFUN=1, but now it will not connect

How did you get this to succeed? Was it at aborting during an upgrade?

eveready1010, 2020-07-09:
I am not 100% sure, but I believe it was because the battery did not have a full charge.  I accidentally left my Arduino on overnight which I had never done in the past, and the next day it finally worked.  So apparently leaving the Arduino on and everything connected overnight allowed the battery to get a full charge which the SIM7000 seems to require.

Tasosgi, 2021-01-18:
I got the same issue. I am using a 3Amp power supply but the CFUN is stuck to 5

WRedux, 2021-03-12:
> 
> 
> Got it resolved, you have to restore xQCN file using QPST tool.

Please: How exactly?

WRedux, 2021-03-12:
Solved it by myself. 
Since I have the LilyGo and not the Botletics-Model, I guess, I have a different Flash-Software.
In mine, the xQCN-Option to parse the xQCN was set to "false" in the .ini of the Flash-Soft (SIM7000 QDL V1.43 Only for Update (backup&recover)), which is called cfg.ini.
Also the Option to ONLY do a xQCN-Flash (and not the complete Firmware) was set to "false".
(Since the Rest of the Firmware was OK, and shows up correctly)
I've set both to "true" and just updated my xQCN and finished up successful.

So my Problem of the not-showing-up IMEI and the Factory-Test (CFUN=5) are History now... :-)
Both appeared after a unsuccessful Firmware-Update (USB Connection got lost while update).