botletics/SIM7000-LTE-Shield #133: SIMCOM7600 cannot connect LTE

hendri-tobing, 2019-09-26:
I am working on SIMCOM7600 with the fona library from this repository. It is stuck on wirelessConnStatus and fona.openWirelessConnection(true). 

[image]

It seems the device does not recognize AT+CNACT command. Please advise.

jupiter8888, 2019-10-06:
Looks like your APN is the problem. Are you sure it is the right one?

felipeandres79, 2019-10-30:
I would suggest read the at command for sim7600, And pdp context parameter seems wrong.

spit4520, 2019-11-26:
the 7600 modem only supports TMobile/Twilio bands, Verizon/AT&T hasn't worked with my experience. I suggest using a Twilio sim for that board, if you have a 7000a/g/e use a Hologram sim. The documentation from SimCom is bad and misleading about this distinction, but I found this out through personal experience

cheers,

Scott