botletics/SIM7000-LTE-Shield #134: readRTC function looks incomplete

zggz, 2019-10-02:
It also doesn't seem to work. I think the code there only parses the year from the response. I plan to update my copy of the library, and if you'd like me to I can try to make a merge request with that version. Am I missing something? Is readRTC working for other people?

botletics, 2019-10-07:
You seem to be correct regarding the readRTC() function. However, please take a look at [the getTime() function](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Library-Functions#fonagettime) which could replace the readRTC() function. You can quite easily parse it in your own code after getting the string of numbers.

botletics, 2020-12-15:
Thanks, I updated the library with [this fix](https://github.com/adafruit/Adafruit_FONA/pull/76)

closed 2020-12-15 by botletics