botletics/SIM7000-LTE-Shield #138: Sim7000 wont send data to IO Adafruit

mauiphill, 2019-11-06:
I have no problems interacting with the LTE Demo (get GPS location, send and receive messages) but I’m having problems with the ioT Example. I can’t send data to IO Adafruit. 
My main aim is to make a GPS unit for my truck. 
Can you please give me some advice on what I should do, I’ve tried for weeks to get it to work. 
Thank you for your time. 
Aloha.

botletics, 2019-11-11:
Could you please attach an AT command log? Please also try out the MQTT demo code.

mauiphill, 2019-11-15:
Hi Timothy. Thanks for your speedy reply. 
I took pictures of my TT command log for the MQTT and the ioT_Example. 
Thank you for your time, I really appreciate it. 

Aloha
Phill
[image]

@
[image]

botletics, 2019-11-18:
Looks like it's perhaps a power issue. Please make sure you have a LiPo battery connected.

mauiphill, 2019-12-02:
The battery is connected and has the correct voltage. 
I’ve looked online for a solution to this problem and I see I’m not the only person with this issue.

botletics, 2019-12-02:
Can you send data to dweet.io using the LTE_Demo and options 'G' and '2'? If so, data is working fine. Then if you still can't post to Adafruit, make sure you read the wiki on how to [set up AdafruitIO so that you can send to it](https://github.com/botletics/SIM7000-LTE-Shield/wiki/GPS-Tracker-Example#adafruit-io)

tjdzor, 2020-01-24:
I originally responded to #127 but its probable most relevant here.

I too had been fighting this issue, but figured a couple things out!

Background, trying to make a cellular temperature logger using AdafruitIO for a vacant property that has had some random heating issues. Using a Hologram sim, everything worked fine in the SIM7XXX_LTE_Demo, but would go into a reboot loop or hang if I tried using the AdafruitIO_MQTT_Demo or IoT_Example. Tried the previously mentioned solutions such as adding a battery, bought a Mega to try thinking maybe it was a stability issue. But alas, no love.

So I added some serial outputs and tried commenting sections out to help figure out where it was failing. I found two issues in setup()

It seems to hang @ tempsensor.wake();
I looked at the Adafruit MCP9808 example and sure enough, theres no wake before initializing it. So I commented out that line. (//tempsensor.wake();)

And next, it would fail to find the MCP9808 anyways. It's being started without the i2c address, so adding the address 0x18 fixed it. The address can be checked with an i2c Scanner. And the sensor can be tested with the Adafruit MCP9808 Library/Example
if (!tempsensor.begin(0x18)) {

I also set the resolution of the temp sensor to the highest (but also slowest), but that probably isn't necessary.
tempsensor.setResolution(3);

Mine has been working flawlessly for a couple days now. Would love to try it on an Uno, but due time constraints its staying on the Mega for the time being.

I really hope this helps some people!

alberg79, 2020-01-25:
I too have this working after commenting out //tempsensor.wake(); thanks for sharing the solution.  I also verified this on the Mega and Uno.  Even though the Uno gives a low memory error it does work.

mauiphill, 2020-01-26:
tjdzor. 
You’re amazing. I’ve literally spent months playing around with the code but had no success, I’d actually given up. 
I commented out the line you mentioned and it’s working perfectly now. 
Thank you soooooooo much.