botletics/SIM7000-LTE-Shield #143: Documentation Error SIM7000 PSM Mode

Thalhammer, 2019-11-17:
I noticed that if you want to use PSM mode from inside a DAM module you NEED to enable "slow clock" in addition to the normal psm setting. Otherwise your module will enter psm mode just fine, but never wake up again.

So the correct procedure to enable PSM should be:
* AT+CSCLK=1
* AT+CPSMS=1

Since it does not harm having the slow clock enabled for non DAM usage I think it is a good idea to include it in the wiki article about power consumption.

botletics, 2019-11-18:
Thanks! Edited the wiki page on power consumption.

closed 2019-11-18 by botletics

jaka87, 2020-12-19:
Did you guys manage to get PSM working ? My network supports it, im able to enter PSM by entering previously mentioned commands (LED stops blinking) but after wake up (pulling pwr pin low) my network connection is lost and it doesn't establish untill i restart module

davegravy, 2021-06-10:
@jaka87 having the same issue. I did you or anyone else solve this?

jaka87, 2021-06-11:
Yes. It was some time ago but i believe the solution was to re-establish serial connection with the module which is dropped while in PSM

codyohl, 2025-12-28:
@jaka87  @davegravy just to confirm, you folks were able to get PSM working? Did it pull the advertised 9uA? Thanks! I’m deciding whether to leave it in PSM or turn it off and on every couple hours

Mark-Wills, 2025-12-28:
Powering the modem down completely will use a LOT more power than PSM mode
in experience, especially in areas with poor reception. 👍

Mark

Thalhammer, 2026-01-05:
@codyohl From the module's power management point of view PSM is essentially identical to power off. The only difference is that it has to "remember" that it is still associated to the Tower and what that association looks like.
When the firmware enteres PSM it sends a message to the cell tower saying "hey, I still like your network but I am going to sleep for the next x seconds" and the network then buffers all messages it has for the module until that timeout expires. The module then turns on its rtc timer, writes the information to the RTC memory (which unlike main ram is kept active during sleep/poweroff) and otherwise shuts down. After the timer expires it powers back on (doing essentially a full boot) and restores the modem state from rtc memory. Then it tells the tower to send everything that happened during its away time. The 2µA difference between PowerOff and PSM is essentially this RTC timer and memory being kept on.

There are a couple caveants though:
- The power savings depend mostly on the sleep period and the period is not arbitrary
- Its mostly useful for stationary things because if you move towers (or worse roam to a different network) it usually need to go through the full reconnect anyway
- It only works in Cat-M and NB-IoT networks. 2G/GSM, which in my experience (sadly) still provides the majority of coverage, does not support PSM/eDRX
- Your milage may vary, depending on the duration you wanna sleep and the network conditions