botletics/SIM7000-LTE-Shield #144: SIM7000E FW Upgrade Fails

madatr, 2019-11-18:
Hello,
Firstly, I'd like to thank you for all of your work and support, you guys are wonderful.

I have been trying to upgrade my SIM7000E Firmware to either B08 or B07 because some commands are not working in the B03 version but I have been running into an issue.

I have tried following both of your guides using the QPST and the QDL update tools with no success.

Trying to upgrade using QPST, after choosing Start the program works for a few seconds i assume (nothing is shown on the log or error sections) then a message pops up saying: "Failed to Enter Sahara" 

Using QDL, the programs runs what i assume are some tests or pre-configurations successfully, ie issuing some AT commands and backing up the QCN but it gets to the "Waiting switch to download mode" and just hangs there indefinitely without returning anything and the progress bar stays at zero percent, i have tried waiting for upto 1000 seconds without it changing so i assume it runs into a similar error as the previous program in entering the download mode.

The only details i can think of providing is that as instructed, I have supplied the module with a 3.7 v source and I have monitored the power LED and it seems that it stays on during the process so my guess is its not a power issue.

Please help.

madatr, 2019-11-18:
If you  don't know/cant help in this regards, please then explain how to upgrade using "FOTA", "Firmware Over The Air" using the command "AT+CFOTA".

What i could understand from the command manual is that the device can self upgrade, providing that the new upgrade file is loaded to the flash memory, but using the example they provided to get the file using HTTP "AT+HTTPTOFS" returns 601 "Network Error", this can be due to many reasons.

Do you know how to load the file to the flash memory of the device and in which format

Thanks,

botletics, 2019-11-18:
According to the [FOTA app note](https://github.com/botletics/SIM7000-LTE-Shield/blob/master/SIM7000%20Documentation/Technical%20Documents/SIM7000%20Series_FOTA_Application%20Note_V1.00.pdf) you use enable data, then use the AT+HTTPTOFS command to open an HTTP get session, then use AT+CFOTA to tell it to grab the file from the link specified in the HTTPTOFS command.

madatr, 2019-11-18:
"AT+HTTPTOFS" returns 601, "Network Error". Is there another method to push the file to the memory and in which format?

botletics, 2019-11-18:
Make sure you enable data (for example, see the "G" command in the LTE_Demo). I'm not exactly sure what you mean by "push the fine to the memory" because the file is on the server, and the module downloads it. It's not already loaded onto the module.

madatr, 2019-11-18:
I mean if I have the file on my computer, how do I upload to the module flash memory directly so I can use the "AT+CFOTA" command.

botletics, 2019-11-18:
You can use the QPST software download tool to put things in the SIM7000's flash memory.

madatr, 2019-11-19:
Ok, so I tried pushing the firmware file from a PC to the /fota/ section using QPST but it does not complete. I think the firmware file is too large as it is the same one used to flash from PC.

Upon examining the FOTA application note for the module, i noticed the following:

> When customer needs to upgrade the onsite SIM7000 series Firmware from Version A to Version B, >they can send the upgrade requirement to SIMCom via their module provider. SIMCom will provide >related **delta firmware file** according to the information of Version A and Version B provided by >customer. The size of the delta firmware file depends on the difference of Version A and Version B.
 
This means that my assumption is correct, the Delta FOTA file is different from the FW one and needs to be pulled from SIMCom.

I tried the link from the application note:
> http://112.74.93.163:12345/1351B05SIM7000E___1351B08SIM7000E/update.zip
but it seems to be offline.

Do you know how can I get the delta firmware file? and did you ever attempt to upgrade the FW using FOTA?

botletics, 2019-11-19:
Hmmm sorry, I don't know.

madatr, 2019-11-19:
Luckily I came across a solution to force the module into download mode.
Here is the link to the post, check the last three answers:
[https://github.com/botletics/SIM7000-LTE-Shield/issues/41](url)

closed 2019-11-19 by madatr

ChasterKras, 2020-02-26:
Url not open(