botletics/SIM7000-LTE-Shield #147: Sim7000A enters boot loop on trying to disable data

spit4520, 2019-11-26:
Hello,

I am currently using your library for my production product. I love the documentation and all of the support around this project. 

I have encountered a issue where modems will randomly enter a state where they cannot reconnect to the cell network and during the while loop for `fona.enableGPRS` I have unplugged the modem and let it sit for a while, I have also replaced the modem with other ones to remedy out hardware issues, but after a while they all end up being plagued with the issue.

The modem is used to connect over MQTT to our cloud and process commands for remote locks. I can strip out application logic and attach my set up and main loop code if that would be helpful

I hope to get some pointers soon on this, I will post debug logs as needed and src

Thanks,

Scott

botletics, 2019-12-02:
What SIM card are you using, and what do you get in response to the following command:

AT+CGREG?

Not being able to connect to the network would be the root of the problem, because if that's the case enabling data won't work.

spit4520, 2019-12-02:
I am using the Hologram simcard. Below is my logic for entering the loop.

The antenna I was using before spontaneously failed so I replaced it with a different one and the issue still persists.

Thanks for your reply

`
int8_r getSignal() {
  // read the RSSI
  uint8_t n = fona.getRSSI();
  int8_t r;

  Serial.print(F("RSSI = ")); Serial.print(n); Serial.print(": ");
  if (n == 0) r = -115;
  if (n == 1) r = -111;
  if (n == 31) r = -52;
  if ((n >= 2) && (n <= 30)) {
    r = map(n, 2, 30, -110, -54);
  }
  Serial.print(r); Serial.println(F(" dBm"));
return r;
}

bool hasService() {
  uint8_t n = fona.getNetworkStatus();
  Serial.print(F("Network status "));
  Serial.print(n);
  if (n == 0) Serial.println(F("Not registered"));
  if (n == 1) Serial.println(F("Registered (home)"));
  if (n == 2) Serial.println(F("Not registered (searching)"));
  if (n == 3) Serial.println(F("Denied"));
  if (n == 4) Serial.println(F("Unknown"));
  if (n == 5) Serial.println(F("Registered roaming"));
  if( n == 1 || n == 5) {
    return true;
  }
  return false;
}

 #if !defined(SIMCOM_3G) && !defined(SIMCOM_7500)
    // Disable data just to make sure it was actually off so that we can turn it on

    while(!hasService()){
      getSignal();
      delay(1000);
    }
    if (!fona.enableGPRS(false)) Serial.println(F("Failed to disable data!"));
    delay(1000);
    // Turn on data
    while (!fona.enableGPRS(true)) {
      Serial.println(F("Failed to enable data, retrying..."));
      delay(2000); // Retry every 2s
    }
    Serial.println(F("Enabled data!"));
  #endif
  mqtt.will(AIO_USERNAME "/feeds/message", "A,0,0");
  mqtt.subscribe(&topic_cmd);
  Serial.println(F("A,0,1"));
`

botletics, 2019-12-03:
Are you using a 3.7V LiPo battery to power the shield?

spit4520, 2019-12-03:
No, I am not using an official shield. I am using [this](https://www.amazon.com/gp/product/B07LCLPPRP/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) for testing. I am powering the MCU and the modem off of my bench top PSU and it should be able to handle the current spikes. The modem was working before with your library with a few pin reassignments. Is the SIMCom modem especially suspect to current spikes?

botletics, 2019-12-03:
Sorry, I don't give support on other hardware.

closed 2019-12-03 by botletics

spit4520, 2019-12-07:
I have bought the Sim7000G version of your shield now and this issue persists. I have been going back and forth with Hologram support for around 2 weeks and the issue still isn't resolved. The G version of your shield seems to connect to the T-Mobile 2G network, when I set the network preference to 38 (LTE only) the board is unable to connect to any cell tower. I am with in half a mile of a cell tower (I can see it outside my window) and I am surrounded by other sim com (sim7500) devices in my office so I know that the network shouldn't be the issue. I have connected a 3.7v battery to the shield and ensured that it has sufficient charge. I have also connected the supplied antenna from Pulse and left everything to sit for a hour or two and nothing reconnected.

OK
at+cops=?

+COPS: (1,"310 260","310 260","310260",0),(1,"310 410","310 410","310410",7),(1,"313 100","313 100","313100",7),(3,"311 480","311 480","311480",7),(3,"311 480","311 480","311480",9),,(0,1,2,3,4),(0,1,2)

I have went through all of the tutorials on hologram's site to clear the black lists [see here](https://support.hologram.io/hc/en-us/articles/360035697373-How-do-I-clear-the-FPLMN-list-?source=search&auth_token=eyJhbGciOiJIUzI1NiJ9.eyJhY2NvdW50X2lkIjo5NDI2MTY2LCJ1c2VyX2lkIjozOTU4MDY5MTc1NzQsInRpY2tldF9pZCI6MTM3OTkxLCJjaGFubmVsX2lkIjo2MywidHlwZSI6IlNFQVJDSCIsImV4cCI6MTU3ODExNDI0Mn0.rWr-UQ5RkhNbMC8wFWkll7Hn01OHFDQOZq8vRHtruIM) and after a reboot the modem now cannot connect to the 2g network either :( I am very confused at this point and really would like some assistance.

`OK
at

OK
ati

SIM7000G R1529

OK
at+cgmi

SIMCOM_Ltd

OK
at+cgmm

SIMCOM_SIM7000G

OK
at+gmm

SIMCOM_SIM7000G

OK
at+cpin?

+CPIN: READY

OK
at+ccid

8944501810180176460f

OK
at+crsm=176,28539,0,0,12

+CRSM: 144,0,"FFFFFFFFFFFFFFFFFFFFFFFF"

OK
at+cfun?

+CFUN: 1

OK
at+csq

+CSQ: 19,99

OK
at+creg?

+CREG: 0,5

OK
at+cgreg?

+CGREG: 0,5

OK
at+cgdcont?

+CGDCONT: 1,"IP","hologram","0.0.0.0",0,0,0,0
+CGDCONT: 13,"IP","hologram","0.0.0.0",0,0,0,0

OK
at+cgact?

+CGACT: 1,0
+CGACT: 13,0

OK
at+cops?

+COPS: 0,0,"310 260 Hologram",0

OK
at+cops=?

+COPS: (1,"310 410","310 410","310410",7),(2,"310 260","310 260","310260",0),(1,"313 100","313 100","313100",7),(1,"311 480","311 480","311480",7),(1,"311 480","311 480","311480",9),,(0,1,2,3,4),(0,1,2)

OK`

botletics, 2019-12-10:
Please do a factory reset with the command "AT&F0" then load the LTE_Demo code and check network connection with "n".

spit4520, 2019-12-11:
FONA> i
	---> AT+CSQ
	<--- +CSQ: 99,99
RSSI = 99: -115 dBm
FONA> i
	---> AT+CSQ
	<--- +CSQ: 99,99
RSSI = 99: -115 dBm
FONA> n
	---> AT+CGREG?
	<--- +CGREG: 0,2
Network status 2: Not registered (searching)
FONA> n
	---> AT+CGREG?
	<--- +CGREG: 0,2
Network status 2: Not registered (searching)
FONA> i
	---> AT+CSQ
	<--- +CSQ: 99,99
RSSI = 99: -115 dBm
FONA> 

I am also using the provided antenna. After my last post I disconnected it being very careful not to damage the UFL connector and placed it in a drawer, I have not ESD'ed onto it and been careful not to bend the edges.

I have a 2500 man battery connected and it is reporting to be at roughly 70% capacity.

I ran the AT&F0 command with ATE enabled and received no error command nor did I receive an OK acknowledgment. 

Any Ideas?

Thank you

botletics, 2019-12-16:
Have you read [this wiki page](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Connecting-to-Network)?

spit4520, 2019-12-18:
CSQ: 99,99

OK

+CSQ: 99,99

OK

OK
ate

OK
ate

OK
ati

SIM7000G R1529

OK
ate

OK
ati

SIM7000G R1529

OK
at+ccid

8944501810180176460f

OK
at+csq

+CSQ: 99,99

OK

NORMAL POWER DOWN
 at

OK
ate

OK
ati

SIM7000G R1529

OK
at+ccid

8944501810180176460f

OK
AT+CGDCONT

OK
AT+CGDCONT?

+CGDCONT: 1,"IP","hologram","0.0.0.0",0,0,0,0
+CGDCONT: 13,"IP","hologram","0.0.0.0",0,0,0,0

OK
at+csq

+CSQ: 99,99

OK
at+csq

+CSQ: 99,99

OK
at+cops?

+COPS: 0

OK
at+cops=?

+COPS: (1,"313 100","313 100","313100",7),(1,"311 480","311 480","311480",7),(1,"310 410","310 410","310410",7),(1,"310 260","310 260","310260",0),(1,"311 480","311 480","311480",9),(1,"310 410","310 410","310410",9),,(0,1,2,3,4),(0,1,2)

OK
at+csq

+CSQ: 99,99

OK
AT+CGMR

Revision:1529B01SIM7000G

OK
at+cnmp?

+CNMP: 2

OK
at+cnmp=?

+CNMP: ((2-Automatic),(13-GSM Only),(38-LTE Only),(51-GSM And LTE Only))

OK
at+cnmp=38

OK
at+cnmp?

+CNMP: 38

OK
at+cmnb=?

+CMNB: ((1-Cat-M),(2-NB-IoT),(3-Cat-M And NB-IoT))

OK
at+cmnb?

+CMNB: 3

OK
at+cmnb=1

OK
at+cbandcfg?

+CBANDCFG: "CAT-M",1,2,3,4,5,8,12,13,18,19,20,26,28,39

+CBANDCFG: "NB-IOT",1,2,3,5,8,12,13,17,18,19,20,26,28

OK
at+cbandcfg=?

+CBANDCFG: (CAT-M,NB-IOT),(1,2,3,4,5,8,12,13,17,18,19,20,26,28,39)

OK
at+cops=?

+COPS: (3,"311 480","311 480","311480",7),(1,"310 260","310 260","310260",0),(3,"313 100","313 100","313100",7),(1,"310 410","310 410","310410",7),,(0,1,2,3,4),(0,1,2)

OK
at+cops=1,3,"313100"

ERROR
at+cops?

+COPS: 0

OK
at+cops=1,3,"311480",7

ERROR

I ran all of the diagnostics from the debug page and after letting it sit for about an hour it eventually reconnected.

My questions stems from if there is any way to speed this process up, or streamline the process? I have really enjoyed all of the support you have given me and enjoy your board.

Best,

Scott

botletics, 2019-12-23:
Please make sure you have a 3.7V LiPo battery plugged in with the right polarity. The red wire should be on the left side of the connector when you plug it in.