botletics/SIM7000-LTE-Shield #152: OPERATOR SELECTION

JGLJGL, 2020-01-24:
Hi, to select an operator in my country I use AT+COPS. But what if I send the SIM7000 module to another country ? Is there any option to connect automatically to the opertator available in that country? So that anyone (with 0 knowledge) could connect the module (once programed).

botletics, 2020-01-24:
You have to know what network you will be on because you need to enter the APN.

ROSW6341, 2020-02-19:
Would the AT+CMCFG command be of use?  Setting the <mode> parameter to 1 would automatically select mobile operator configuration according to ICCID information in SIM card.  Other options are also available.

botletics, 2020-03-09:
I believe module will automatically try to connect to the best available network in the area based on the SIM card that's inserted and the networks nearby. If you use the AT+CNMP and AT+CMNB commands you can narrow down the type of network it tries to connect to, then use AT+COPS=0 to make it search automatically.

closed 2020-03-10 by JGLJGL