botletics/SIM7000-LTE-Shield #153: sim7000a w/ hologrsm sim not working

kylekutscher, 2020-01-24:
Hello, 
  I recently purchased a sim7000a and a hologram sim. I am having a problem with getting the networking activated. I load the esp32 lte demo, leaving everything the same and connecting only the pins required(RX TX GND 3.3) select "G" and its giving me "AT+SAPBR=1,1 error".
   I'm also having an issue keeping the board to stay on, it will randomly turn off. im using a brand new Lipo battery and the polarity is correct, I also have the sim7000a running off a power supply with common grounds to the esp. 
 Can someone direct me in the right path to get this up and running?

botletics, 2020-01-25:
Could you please provide exactly what each pin on the SIM7000 board is connected to? It sounds like you somehow have a power issue. You should only plug in the LiPo battery to the JST connector (or VBAT and GND) pins and if you want you can charge the battery with the ESP board.

kylekutscher, 2020-01-26:
fona(10)->esp(16)
fona(11)->esp(17)
fona(7)->esp(5)
fona(6)->esp(18)
I cut the trace and soldered the jumper for IO.
logic vcc -> esp 3.3v
lipo plugged in. 
5v -> power supply
common grounds

botletics, 2020-01-26:
If you didn't make the solder jumper modification with a razor blade then you should connect the 3.3V logic to the SIM7000 board's 5V pin actually. The 5V pin won't properly power the shield alone (you need the LiPo battery plugged in) and it only charges the battery. If you need to charge the battery without making the solder jumper modification you can just charge the battery from a different board (for example, the Adafruit Feather boards have LiPo battery charging). Please see [this wiki page](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Pinouts) which explains the power and logic pins.

kylekutscher, 2020-01-27:
I did cut the trace with a blade and soldered the jumper there. I connected 3.3v to the logic vcc labeled  near the jumper as well as ground. then with a power supply, 5v to input labeled "5v" since I cut the trade.

botletics, 2020-01-28:
Does it turn off only when you try to enable data, or does it turn off simply being connected to the network? Does it turn off by itself without a SIM card inserted?

kylekutscher, 2020-01-28:
I havent got it connected to the network yet, that is part 2 of my issue im having. It does the same with or without the sim inserted.

botletics, 2020-01-29:
Hmmm. Have you tried just powering the shield alone with only the LiPo battery (and without the Arduino attached)? It shouldn't be powering off on its own with a sufficiently large LiPo battery (1000mAH+ capacity)