botletics/SIM7000-LTE-Shield #155: Slow MQTT Publish

Martin-FO, 2020-01-26:
Hi
Thanks for a great product, library and examples. It really helps a lot.
But I have a question: When using MQTT and publishing to CloudMQTT it takes 20-30sec for each publish, which seems like a lot to me. 
Sometimes I also get a "failed error" even though CloudMQTT received the data. I think it is a timeout issue, since the "OK" wasn't received in time. 
PS. I am using the NB-IOT network.

It this just to expect or is there something wrong and/or something I can do to improve it?

botletics, 2020-01-26:
NB-IoT can have high latency times but I haven't had the opportunity to test NB-IoT myself.

jaka87, 2021-01-06:
I had the same issue. The problem is not with the module nor the network but in the code. 
In 
`Adafruit_FONA_LTE::MQTT_publish`  
getReply is send mutch to high (20 seconds), while it cand be set much lower, I have tested with 300ms and so far had no problems...
`  getReply(cmdStr, 300);
`

botletics, 2021-01-06:
Thanks for the note, I changed it to 2000 instead of 20000, which I agree is much too long.