botletics/SIM7000-LTE-Shield #161: HTTPS

JGLJGL, 2020-02-14:
Hi, 
I know this code is for HTTP GET and POST, but is it posible to do HTTPS GET and POST using this code ?

jackattack21, 2020-07-01:
many people would like to know the answer to this @botletics can you help us

kartemann, 2020-08-17:
Has anybody made progress on this? @botletics, this would be a huge help in getting our project off the ground, deploying hundreds of devices, with your hardware hopefully.  

Thanks,

Humancell, 2020-08-18:
Have you read through the app note from SIMCom?

https://www.waveshare.com/w/upload/5/5c/SIM7000_Series_HTTPS_Application_Note_V1.01.pdf

I'm not sure the various calls have been integrated into the library yet.  I might look at doing this as I'm doing some testing myself right now.

botletics, 2020-08-20:
@JGLJGL used [these AT commands](https://github.com/botletics/SIM7000-LTE-Shield/issues/165#issue-568893959) which worked for him.

kartemann, 2020-08-20:
Thanks for the comment, what's the best way to incorporate the AT commands into an to sim7000 IOT_example sketch to make an https POST?

botletics, 2020-08-24:
You can use the "sendCheckReply" command to send AT commands in your sketch for quick testing.