botletics/SIM7000-LTE-Shield #163: Verizon network continually on FPLMN list (network forbidden, network registration denied)

alberg79, 2020-02-18:
Hello I am successfully using a botletics SIM7000A  v6 shield with a hologram sim to connect to AT&T's network (band 12).
 
My problem comes when trying to use the same hardware on Verizon's network.
I have used the AT command to lock to band 13 only for CAT-M but registration on the Verizon network keeps getting denied. The Verizon network is getting added to the FPLMN list. If cleared from the FPLMN list it is added again within about a minute or less. 

I have tried the AT command to manual lock to 311480 but this did not help.
I have tried waiting overnight for registration and again this does not work. I suspect once a network is on the FPLMN list it will not try registering any longer no matter how long you wait.
I have tried 5 different Verizon cell towers.
I have tried in a location with max signal strength (CSQ 30).  
I have tried two different shields. 
I have followed all troubleshooting steps I can find online. 

My hologram device activity only shows the device connecting on AT&T never on Verizon.

Here are the responses to the command on the hologram network connection troubleshooting page.

Thanks in advance for any suggestions you may have.

FONA> S
Creating SERIAL TUBE

OK

STATE: IP INITIAL

OK
AT+CIPSHUT

SHUT OK
AT+CIPSTATUS

OK

STATE: IP INITIAL
AT+CIPSTART="TCP","cloudsocket.hologram.io",9999

OK

STATE: PDP DEACT

CONNECT FAIL
AT+CSTT="hologram"

ERROR
AT+CIICR

ERROR
AT+CIFSR

ERROR
AT+CIPSTART="TCP","cloudsocket.hologram.io",9999

ERROR
AT

OK
ATI

SIM7000A R1351

OK
AT+CGMI

SIMCOM_Ltd

OK
AT+CGMM

SIMCOM_SIM7000A

OK
AT+GMM

SIMCOM_SIM7000A

OK
AT+CPIN?

+CPIN: READY

OK
AT+CCID

894450xxxxxxxxxxxx9f  (x's are in place of my sim number, but matches my hologram number with the exception of the "f" at the end - apparently normal to have an additional 20th digit added when issuing the CCID command from what I see online)

OK
AT+CRSM=176,28539,0,0,12

+CRSM: 144,0,"130184FFFFFFFFFFFFFFFFFF"

OK
AT+CFUN?

+CFUN: 1

OK
AT+CSQ

+CSQ: 15,99

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CREG?

+CREG: 0,2

OK
AT+CGDCONT?

+CGDCONT: 1,"IP","hologram","0.0.0.0",0,0,0,0
+CGDCONT: 13,"IP","CMNET","0.0.0.0",0,0,0,0
+CGDCONT: 2,"IPV4V6","ims","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,0
+CGDCONT: 3,"IPV4V6","sos","0.0.0.0.0.0.0.0.0.0.0.0.0.0.0.0",0,0,0,1

OK
AT+CGACT?

+CGACT: 1,0
+CGACT: 13,0
+CGACT: 2,0
+CGACT: 3,0

OK
AT+COPS?

+COPS: 0

OK
AT+COPS=?

+COPS: (3,"Verizon","Verizon","311480",7),,(0,1,2,3,4),(0,1,2)

Humancell, 2020-02-18:
Hello,

I can't promise that this is the issue here, but in the past I found that Verizon will not allow *ANY* device on their network unless it is device carrier certified.  I've had numerous "certified modules" that I have tried to connect to the Verizon network, and in each case was told that "module certification" is not good enough ... you must have end device certification to be allowed to connect.

The last time I tried was using a Quectel EC25-V module on a prototyping board ... the module is Verizon Certified ... but my Verizon rep explained that since it was not associated with a certified device it would never work.

alberg79, 2020-02-20:
Thanks Humancell for your response.  I did speak with Hologram support and received this response.
"Hologram has access to Verizon, but they are being extremely stringent on who we can and cannot give access to. Because of this, at the moment we are only able to provide Verizon access to certain large deployments.

All devices on the Hologram network should be set to automatic carrier selection (at+cops=0) in order to work properly. Moving forward please make sure to leave that setting on automatic"

alberg79, 2020-02-20:
If someone knows of a way to get a single working Verizon sim for CAT-M or NB-Iot please let me know.

Humancell, 2020-02-20:
Again, I've fought this same battle and completely dropped trying to "experiment" on Verizon.  It's not about the SIM, but their network is going to reject any device that is not fully certified - by Verizon - as a device.  You could reach out and call Verizon directly and discuss this, but I've only run into dead ends.  If you *do* find a solution please post it here ... I'd love to find a way around this.

Also, T-Mobile is also saying they are going to do the same for NB-IoT devices, if you have more than a handful.  Right now they are not restricting the NB-IoT devices I'm using on their network.  AT&T also seems to allow them on.

botletics, 2020-02-24:
Verizon is being strict with what devices they let onto the network and yes, I think they recently went through and stripped network priveleges for a lot of hardware that's not Verizon certified.

aemoreno1982, 2020-02-24:
I can confirm the above.  My modules used to continually connect up to Verizon, but suddenly preferred AT&T Mobile as of late.

cbrake, 2020-03-06:
I think the issue is using a MVNO SIM on Verizon for Cat-M devices. If you use a Verizon SIM, it will connect, even in an uncertified device. With the Verizon thingspace developer plan, you can activate up to 20 devices at a very reasonable price: 

https://thingspace.verizon.com/service/connectivity/

Beyond that, you need to be in their M2M (expensive), or partner program (hard to get in). Another alternative is to use https://www.kajeet.net/ -- they provide competitive plans using Verizon SIMs.

aemoreno1982, 2020-03-06:
Thanks for the info and link, got a batch of Verizon sims on the way.  Cheers!

cbrake, 2020-03-06:
@aemoreno1982 NP, let us know how it goes.

K2AAE, 2020-03-08:
@cbrake Damn! I thought I had found something nobody else had found: Thingspace! After beating the keyboard to death Hologram informed me of the Verizon issue (after I'd already order a batch of new cards thinking that was the issue), which for me is a show stopper since AT&T and all others don't work around here. I signed up but couldn't find where to 'purchase' the SIM cards at so I contacted someone at Verizon, they wanted to know how I found out about them - Google Verizon IOT - then directed me to where I could order the SIM. I got into the PO BOX crap with them (rural folk that live far off the beaten path do not get mail delivered, they are assigned Post Office Box and marked from the USPS mailing address list as non-deliverable or INVALID Street Address) Verizon can't send to PO Boxes and my address shows it isn't valid.
All fixed. I have 10 coming for existing boards of which my Botletics board will not let me upgrade the firmware, another post.

botletics, 2020-03-09:
So you're saying it connects with the Thingspace SIM card?

K2AAE, 2020-03-12:
NOPE, They will take your money and set you up with a plan, send you 20 SIM cards but when you try to activate the cards you discover you have to get a "certified" IMEI device, botletics is not a 'certified' device nor is the other 6 LTE M1 devices I have. In fact a device that is on Verizon's site that says it is a certified device failed. When I ask about it I was told I had to get 'my product' (I added a power source) certified, sign  a NDA, provide proof of an Insurance Bond, have the product tested by a third party then submit the paper work and money then wait.

Anyone want one of these useless Verizon SIM cards?

cbrake, 2020-03-12:
I may have been wrong about this -- I've only used verizon thingspace with BG96 Quectel modems. The modem is certified for Verizon, but not end device certified (VZ still requires you to end device certify a product that may use the BG96) -- so perhaps there are two levels of certification required? This cellular stuff is complex ...

K2AAE, 2020-03-12:
This is my response to the BG96 vendor:
I could not get the Hologram SIM to connect to Verizon (the only carrier in my area) so I signed up with https://thingspace.verizon.com/service/connectivity/ (Typical, broken, badly thought out, user unfriendly Verizon site) and ordered 20 SIM cards. They arrived yesterday, when I tried to activate the card it would not accept the board/device IMEI it said the IMEI was not in the DLD(? forgot already) database, at any rate I could not activate a SIM. I sent the link to Verizon showing the  product on their 'Certified Product' page and was told I had to:

    Submit life history
    Provide Proof of insurance bond
    Sign NDA
    Have Third Party Compliance Testing done
    Submit the results of the above and wait, no money was mentioned but I'm sure there is some.

As far as I can tell, it is called generating a new money source. The module I used was a certified MODEM, I send AT commands and they want my end product? There are two other products on the Verizon certified list I'm checking out.

Humancell, 2020-03-12:
Hello,

I tried to explain this before ... and you will only continue to be frustrated with Verizon.  There is a HUGE difference to them between a "Certified Modem" and a "Certified Product".  Your product MUST contain a "Certified Modem" ... and then your PRODUCT must be certified also.  They suck.

I've gone down this road with Verizon and then given up.  They insist that the final complete PRODUCT be certified, and the IMEI assigned to that product must be registered in their system.  The MODEM must also be certified.  They suck.

Go to AT&T or T-Mobile and you will have a lot less stress.  And check out a Twilio SIM.  :-)

Humancell, 2020-03-12:
Also ... above you wrote:  "... botletics is not a 'certified' device nor is the other 6 LTE M1 devices I have."

The cellular MODULE that is on or in these devices might be a "Certified Modem/Product" (and the SIM7000x IS certified) but the final product is not certified, and the IMEI is not registered with Verizon.

Verizon does not trust that you will interact with the modem and their network in a way they like ... and they want your money!

cbrake, 2020-03-12:
I was told the point of the Verizon Thingspace program is to give you access to the network for a limited # of devices so you can develop your product before it is end-device certified. So a Quectel BG96 modem will not work before the product is end device certified?

I should clarify -- my tests were done with a Nimbelink BG96 modem module, which is end device certified with Verizon.

K2AAE, 2020-03-13:
"I should clarify -- my tests were done with a Nimbelink BG96 modem module, which is end device certified with Verizon."
Nimbelink is ODI compliment, look at the datasheet you will find Verizon ODI on it. I spoke with Mike Tackaberry,  Vice President Sales at Nimbelink about this issue, their 'modem'/module are ODI, plug them in and register the SIM. 
I can see having to re-certify if you have modified the firmware, there is an ARM 7 inside to play with and the vendors are giving out the SDK, BUT to have to spend the time and money to re-certify an unaltered module/modem is insane.

K2AAE, 2020-03-13:
In Quectel BG96 datasheet:
5.1 Carrier Specific CertificationsNL-SW-LTE-QBG96:​ Verizon ODI, PTCRB, AT&T, GCF, Telus, Orange
NOTE: at the Verizon site https://opendevelopment.verizonwireless.com/device-showcase/device/7512 it does not list that this device is ODI compliment - wonder why?

ALEEF02, 2020-11-10:
I'm new to Hologram SIM and am trying to register with Verizon. What I've gotten at from this issue thread is that it is not possible because they are restricting to only Verizon certified hardware, correct?

jwallis, 2021-04-01:
"I'm new to Hologram SIM and am trying to register with Verizon."

Hologram SIM with the Botletics shield will probably connect AT&T or T-Mobile but not Verizon.  Not because of the SIM but because the the Botletics shield is not certified.

Basically, don't try to register anything developer-related with Verizon.

jwallis, 2021-04-01:
@botletics this can be closed

closed 2021-04-01 by botletics