botletics/SIM7000-LTE-Shield #164: Some Fixes

ROSW6341, 2020-02-19:
I'm new to posting enhancements/corrections to github so please bear with me.  In the course of working with the library using a SIM7000A with an ESP32 a couple of things showed up.  First there is no UDP support.  It was easy to add, so a working version is in the attached patch file along with an example INO showing its usage to get a timestamp from an NTP server.  Second, there several problems getting GPS information when only some of the available data was needed.  It has only been tested with the SIM7000 but should be good with other modules.  That is also in the patch file.  Third, the getRSSI() function was modified so it returned a -1 on error.  The current version returns a 0 for an error which conflicts with a valid return value.  I've also added a method to return the module's RTC values in a struct in addition to the existing discrete method.

NOTE:  The code has only been tested with a SIM7000A & ESP32.

[SIM7000-PatchCode.txt](https://github.com/botletics/SIM7000-LTE-Shield/files/4227215/SIM7000-PatchCode.txt)

 values.

aemoreno1982, 2020-02-24:
Thanks for posting this!  The UDP support is very handy.

ROSW6341, 2020-03-02:
I discovered an error in the NTP routine in the 'ESP32_UDP_NTP_Demo.ino' example code.  If you are using it, the
    (unsigned long)(timezone * SECS_PER_HOUR)
on line 403 should be changed to
    (long)(timezone * SECS_PER_HOUR)

closed 2020-03-02 by ROSW6341