botletics/SIM7000-LTE-Shield #165: SIM7000 HTTPS  works at "AT Command Tester" but not in "Arduino"

JGLJGL, 2020-02-21:
I made to do both GET and POST in HTTPS, with the programm "AT Command Tester".

But not being able to achieve them if trying to send the comands manually for example with "RealTerm" or with a bridge sketch in "Arduino". I get this responsens por GET and POST int HTTPS:
+SHREQ: "POST",400,800
+SHREQ: "GET",400,800

Any idea why it works in one application and not in others?

Here I add the commands I send in case it is helpfull (POST case):
`AT+SHDISC
AT+CGREG?

AT+SAPBR=3,1,"APN","ep.inetd.gdsp"

AT+SAPBR=1,1
AT+CNACT=1,"ep.inetd.gdsp"

AT+SAPBR=2,1
AT+CNACT?

AT+CFSTERM
AT+CFSINIT
AT+CSSLCFG="convert",2,"rapidssl.cer"
AT+SHSSL=1,"rapidssl.cer"

AT+SHCONF="URL","https://www.atesens.com"
AT+SHCONF="BODYLEN",1024
AT+SHCONF="HEADERLEN",350
AT+SHCONF="IPVER",0
AT+SHCONF="TIMEOUT",30

AT+SHCONN

AT+SHAHEAD="Host:","www.atesens.com"
AT+SHAHEAD="Accept:","*/*"
AT+SHAHEAD="Content-Type:","application/json"
AT+SHAHEAD="User-Agent:","Opera"

AT+SHBOD="{'PM1':4.0}",11

AT+SHREQ="/api/v1/4444/telemetry",3

AT+SHDISC`

botletics, 2020-03-09:
Perhaps it's a line ending issue? So you're saying those commands above work perfectly in AT Command Tester but not in RealTerm?

JGLJGL, 2020-03-10:
Yes, that happened. Finally  made it work. Headders problem, I commented "Host" and "Content-Type" and added "Connection:close" and it started t work perfectly.

Bal-Ben, 2021-05-29:
Does anyone really managed to make https work or is it just a myth? :)

davegravy, 2021-05-31:
@Bal-Ben I got it to work for a GET request to `https://httpbin.org/anything`.

HTTPS and MQTTS to AWS IoT is not working for me however.

DarshanDodal, 2022-03-09:
@davegravy have to managed to make MQTTS work?.

rhr407, 2022-04-18:
Checking registration status...

AT+CREG?

+CREG: 2,1,"912","3D73",0

OK
The device is registered in home network.

AT+CGREG?

+CGREG: 2,1,"912","3D73",0,"1"

OK
The device is registered in home network.

Device is registered.. 

Check the network APN...

AT+CGNAPN

+CGNAPN: 0,""

OK
Network did not send APN to the device.
Activate the network bearer...

AT+CNACT=0,1

OK

+APP PDP: 0,ACTIVE
Set up the HTTP URL...

AT+SHCONF="URL","httpbin.org"

OK
Set up the HTTP body length...

AT+SHCONF="BODYLEN",1024

OK
Set up the HTTP header length...

AT+SHCONF="HEADERLEN",350

OK
Initiating HTTP connection...

AT+SHCONN

OK
Get the HTTP connection state...

AT+SHSTATE?

+SHSTATE: 1

OK
HTTP connection is successful...

HTTP get request...

AT+SHREQ="http://httpbin.org/get",1

OK
No reponse received..

ZakiOro, 2022-06-01:
Is there any update on this?

Scrts, 2022-07-09:
> @davegravy have to managed to make MQTTS work?.

Hi, I've managed to get MQTT work. Follow @davegravy flow [here](https://github.com/botletics/SIM7000-LTE-Shield/issues/58#issuecomment-851797623)
However, what was important for me was that the module starts with time 1980-Jan-01, so AWS rejects the connection. I had to do time update manually (AT+CCLK) to test, but eventually setting up NTP for automatic time sync works:
AT+CNTP="pool.ntp.org",-16,0,0 (this is for EST time zone).

Once time was correct - AT+SMCONN returned OK and I could do subscribing and publishing with no problems.

botletics, 2022-12-04:
For HTTPS, please [see this info](https://github.com/botletics/SIM7000-LTE-Shield/issues/156#issuecomment-1335989351)