botletics/SIM7000-LTE-Shield #166: Sim 7000g intermittent cellular signal

brownjug, 2020-02-23:
I have had 3 modules and on my 4 th ,,   Everyone of the modules experience a loss of signal periodically , and are not dependable to be there when needed,   I am trying to adapt another antenna on the LTE connection, but I haven’t secured a proper signal with anything I try,,   I noted that I cannot receive or transmit anything when the signal is above (-98 dbm). Do you have any suggestions on a better antenna!!  Is the arduino shield not powerful enough to operate a higher gain antenna?

botletics, 2020-02-24:
What SIM card are you using? Make sure the SIM supports LTE CAT-M, not just traditional 4G LTE like in most phones. The antenna shouldn't be a problem, and LTE CAT-M has better building wall penetration than traditional 4G LTE.

brownjug, 2020-02-24:
Thx ,   Yes I have the hologram SIM, I have been developing my own Gps
tracker program for vehicles, but I always go back to yourLte 7000 example
to ensure proper commands and such.  It doesn’t matter whether I use your
program or mine I have the same problem!   Today I rigged an old satellite
dish (24 inch’s). And installed the antenna as you  would a LNB for TV and
was able to achieve a stable signal,  pointed directly at the cell tower
about 1/2 mile away!  That’s not what I really want to do!!  I also made an
extension cable for the supplied antenna and tried remotely mounting ,,
 that didn’t work.  The loss’s were about 20 dB on a 10 ft extension,   I
was thinking of using a a bigger coax,  maybe a RG59 or RG 6 even.  Would
that help the signal??  Thx for getting back to me,  I want to mount this
unit in a small chassis and attach to ignition system on a vehicle!   And
need dependable remote operation??

ahuangJM, 2020-02-24:
I am experiencing the same problem as well.

I am currently using SIM7000G and using Rogers Canada SIM card. I don't yet know how to find out if their SIM card supports LTE CAT-M or not, but I will figure it out today. Perhaps by calling them.

I didn't do a detailed signal measurement like @brownjug to see what's wrong. But I have noticed that the shield often loses service.

I am collecting data from weather stations and sending the data back via. text. The stations lose services quite frequently and for a long time.

I changed the service APN to "ltemobile.apn" instead of "hologram" used in the example codes.
https://www.google.com/search?q=Roger+apn&oq=Roger+apn

An example of the service lost is:
[image]

ejri, 2021-03-15:
@ahuangJM have you been able to connect to rogers' LTE-M? 

I'm trying this combination of Rogers' APN, and LTE Cat-M, but it doesn't seem to connect. 

// rogers
const char apn[] = "ltemobile.apn";

```
      //2 Auto
      // 13 GSM only
      // 38 LTE only
      // 51 GSM and LTE only
      modem.setNetworkMode(38);
      delay(500);

      /*
          1 CAT-M
          2 NB-Iot
          3 CAT-M and NB-IoT
        * * */

      modem.setPreferredMode(1);
      
``` 

I keep getting:

```
OK
AT+CGREG?
AT+CGREG?

+CGREG: 0,2
```

What did you connect to eventually?

brownjug, 2022-01-25:
well now  2 yrs later I still haven't had success with my 7000G's    Is there a process to upgrade the firmware,,, I also got a new computer of late,,, It doesn't want to acknowledge the ADAFRUIT FONA library commands,, "type fona does not name a type"

i will keep searching for solutions!!