botletics/SIM7000-LTE-Shield #168: FTP doesn't work

benjamin0906, 2020-02-25:
Hi,
I have to set up a FTP connection with a SIM7000 modul, but unfortunatelly I cannot do it and I don't know what is the problem. I attached the log of the GSM commands, as you can see I always got a error 66 after trying to connect to the server.
Would you have any idea what i do wrong?
Thanks!

The 90.130.70.73 IP address is the ftp://speedtest.tele2.net/ test server.
[ftp log.txt](https://github.com/botletics/SIM7000-LTE-Shield/files/4250258/ftp.log.txt)

Humancell, 2020-02-26:
There are a couple of good references:

https://m2msupport.net/m2msupport/tutorial-for-simcom-m2m-modules/#Ftp

https://m2msupport.net/m2msupport/atftpget-set-download-file/

On the second one I found:  "66 operation not allow"

Have you tried to do this same get with an FTP client application and it works?

benjamin0906, 2020-02-26:
Unfortunately I can't see any new information on the sites you linked in.
Yes I've tried to reach the server with various FTP clients and all of them worked.

I made a little mistake: I did not wait for the +FTPGET: answer and sent a new FTPGET=1 command to the module before that and this coused the "66 opertion not allow" error. So the real error code is the "63 Connect error".
But I still don't know what is the promblem.

Humancell, 2020-02-28:
You could always download and test with this tool:  https://m2msupport.net/m2msupport/download-at-command-tester-for-simcom-modules/

It is free for 7 days, and very much worth buying after that.  It has the FTP command, and you can see the exact command it issues.

benjamin0906, 2020-03-04:
Thank you!
I found out what caused the problem. I changed the sim card and it works now. The previous one was a roaming SIM card.

pcbtborges, 2020-05-14:
Hi, I tried to use FTP but compilation stops at:

Line 92:  Adafruit_FONA_LTE fona = Adafruit_FONA_LTE();

'Adafruit_FONA_LTE' does not name a type

Thanks in advance
Paulo

botletics, 2020-05-15:
Looks like you need to follow [these steps](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Arduino-IDE-Setup) to get the proper library installed.

danielehrhardt, 2020-10-31:
How large you the Files be with the FTP Protocol?