botletics/SIM7000-LTE-Shield #173: 7000g

Naoki95957, 2020-03-22:
Hello! I've done everything about as cookie-cutter as the wiki has explained it. I'm a bit confused based on what's in-front of me however...
Hardware:
2Ah Lipo
Hologram.io sim
gps/lte antenna hooked up 
shield attached to arduino
usb from my machine to arduino

The result is a blinking net led, which would appear communication is operating correctly, right? its blinking about twice per second. 

When I launch the

botletics, 2020-03-23:
The network (blue) LED blinking quickly indicates it's not connected, unless it's blinking very rapidly, which means it's connected and data is activated (but I don't think that's your case).

Please see [this wiki page](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Connecting-to-Network) if you're having problems connecting to a network.

pmethods, 2020-03-23:
FYI I validated the botletics branch of the FONA library against a SIM5320A.  It works exactly as described and the sections (2) and (3) are a perfect starting example for doing generic HTTP interfacing (for roll your own IoT)

I extended the library out to (4), (5), (6) for pulling a timestamp down from AWS, Generatring Version 4 signing certificates, and performing an arbitrary file upload to Amazon S3.

botletics wrote a very thoughtful interface that is compatible with a wide range of hardware sets, including stuff offered by other vendors.  If you are having troubles it is ok. . .   this is complicated stuff with a lot of details.  Rest assured that your hardware and software will work - you just have to keep focused on the details and work hard to understand all of it.

Good Vendor to do business with!
Gets my thumbs up.

-methods

Naoki95957, 2020-03-23:
I appreciate the swift replies! I'll dig thru the wiki some and try to figure it out.. I haven't had a lot of time to play with the unit and the initial one I messed with (7000A) worked out of the box. I had a few things land on me recently but I'm very eager to work with these assets and this product! I'll report back and consider this closed unless I come across another roadblock.

closed 2020-03-23 by Naoki95957

Naoki95957, 2020-03-30:
Yay, so this link helped me. I couldn't get the 'serial tube' from the Arduino sketch to work, but after getting a micro usb and installing the windows drivers, I was able to do some AT commands which resolved the issue. Particularly after setting the '+CNMP' and '+CNMB' to automatic/both options. Thanks!

> Please see [this wiki page](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Connecting-to-Network) if you're having problems connecting to a network.