botletics/SIM7000-LTE-Shield #174: More details when writing to a Raspberry Pi

ksaye, 2020-04-02:
I have read https://github.com/botletics/SIM7000-LTE-Shield/wiki/Raspberry-Pi and https://github.com/botletics/SIM7000-LTE-Shield/wiki/Pinouts.  Both contain great information.  

I would like to wire this to a Raspberry Pi 3, and would not like to use the USB.  If I understand the 'Raspberry-Pi' page, it is using it as a UART, so I only need TX, RX, GPIO (to pulse), 5V and GND from the Pi to the shield, and I am also using the LiPo battery that I want to charge from the Pi.

This being the case, would my wiring be as follows?

Purpose | Pi 3 | Botletics SIM7000A
------------ | ------------ | -------------
Pi 5v TX to shield | pin # 8 (TXD0) | D10
Pi 5v RX from shield | pin # 10 (RXD0) | D11
Pi to turn on shield | pin # 12 (GPIO18) | D6
Pi 5v power to shield | pin # 4 (+5V) |  J1 pin 4 (5v/Logic)
Pi Ground to shield | pin # 6 (GND) |  J1 pin 3 (GND)

In my startup, I need to pulse on the Pi pin # 12 to turn the shield on.

Then if i enable the UART on the Pi (/boot/config.txt enable_uart=1), then what change do I make to get the QMI modem to look to /dev/ttyS0 vs the blog which says /dev/ttyUSB2?

Are there any modifications I need to do to my v6 shield and will this charge the battery?

ksaye, 2020-04-03:
@botletics

botletics, 2020-04-03:
Looks like you swapped TX and RX there. Pin D10 is the shield's TX which should be the RPi's RX, and D11 is the shield's RX which should be the RPi's TX.

botletics, 2020-04-03:
Unfortunately I'm not really a RPi user lol. Perhaps someone else can answer your RPi questions.