botletics/SIM7000-LTE-Shield #175: wwan0 error NO-CARRIER on Raspbery Pi

ksaye, 2020-04-03:
I am struggling with getting the shield working on a PI.  I am using a Hologram SIM, and it works in my Windows device (gets an IP address).

I have installed using [https://github.com/botletics/SIM7000-LTE-Shield/wiki/Raspberry-Pi](https://github.com/botletics/SIM7000-LTE-Shield/wiki/Raspberry-Pi) and it installed correctly.

Here is my config file

> root@raspberrypi:/home/pi# cat /etc/qmi-network.conf
> APN=hologram,ip-type=4

Here the pi does see the SIM

> root@raspberrypi:/home/pi# gammu identify
> Device               : /dev/ttyUSB2
> Manufacturer         : SIMCOM_Ltd
> Model                : unknown (SIMCOM_SIM7000A)
> Firmware             : Revision:1351B03SIM7000A
> IMEI                 : 865235030310854
> SIM IMSI             : 234500007021962

raw_ip = Y
> root@raspberrypi:/home/pi# cat /sys/class/net/wwan0/qmi/raw_ip
> Y

running the qmicli

> root@raspberrypi:/home/pi# qmicli -d /dev/cdc-wdm0 --device-open-"net=net-raw-ip|net-no-qos-header" --wds-start-network="apn=hologram,ip-type=4" --client-no-release-cid
> [/dev/cdc-wdm0] Network started
>         Packet data handle: '2197542992'
> [/dev/cdc-wdm0] Client ID not released:
>         Service: 'wds'
>             CID: '1'
> 

yet no carrier:

> root@raspberrypi:/home/pi# ip a
> 1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
>     link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
>     inet 127.0.0.1/8 scope host lo
>        valid_lft forever preferred_lft forever
>     inet6 ::1/128 scope host
>        valid_lft forever preferred_lft forever
> 2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
>     link/ether b8:27:eb:d9:10:0f brd ff:ff:ff:ff:ff:ff
> 3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
>     link/ether b8:27:eb:8c:45:5a brd ff:ff:ff:ff:ff:ff
>     inet 192.168.15.31/24 brd 192.168.15.255 scope global wlan0
>        valid_lft forever preferred_lft forever
>     inet6 fe80::ee98:15ae:f5d5:f556/64 scope link
>        valid_lft forever preferred_lft forever
> 9: wwan0: <NO-CARRIER,POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state DORMANT group default qlen 1000
>     link/none
> 

OS version:

> root@raspberrypi:/home/pi# uname -a
> Linux raspberrypi 4.14.52-v7+ #1123 SMP Wed Jun 27 17:35:49 BST 2018 armv7l GNU/Linux

Here we see dmesg:

> [ 4751.556849] usb 1-1.3: new high-speed USB device number 9 using dwc_otg
> [ 4751.688916] usb 1-1.3: New USB device found, idVendor=1e0e, idProduct=9001
> [ 4751.688930] usb 1-1.3: New USB device strings: Mfr=3, Product=2, SerialNumber=4
> [ 4751.688938] usb 1-1.3: Product: SimTech SIM7000
> [ 4751.688947] usb 1-1.3: Manufacturer: SimTech, Incorporated
> [ 4751.688955] usb 1-1.3: SerialNumber: 1234567890ABCDEF
> [ 4751.692951] option 1-1.3:1.0: GSM modem (1-port) converter detected
> [ 4751.693361] usb 1-1.3: GSM modem (1-port) converter now attached to ttyUSB0
> [ 4751.694145] option 1-1.3:1.1: GSM modem (1-port) converter detected
> [ 4751.697205] usb 1-1.3: GSM modem (1-port) converter now attached to ttyUSB1
> [ 4751.698046] option 1-1.3:1.2: GSM modem (1-port) converter detected
> [ 4751.698394] usb 1-1.3: GSM modem (1-port) converter now attached to ttyUSB2
> [ 4751.699171] option 1-1.3:1.3: GSM modem (1-port) converter detected
> [ 4751.699482] usb 1-1.3: GSM modem (1-port) converter now attached to ttyUSB3
> [ 4751.700299] option 1-1.3:1.4: GSM modem (1-port) converter detected
> [ 4751.700605] usb 1-1.3: GSM modem (1-port) converter now attached to ttyUSB4
> [ 4751.707560] qmi_wwan 1-1.3:1.5: cdc-wdm0: USB WDM device
> [ 4751.708459] qmi_wwan 1-1.3:1.5 wwan0: register 'qmi_wwan' at usb-3f980000.usb-1.3, WWAN/QMI device, a2:4d:e9:a5:3c:3c
> [ 4881.009163] IPv6: ADDRCONF(NETDEV_UP): wwan0: link is not ready

closed 2020-04-23 by ksaye

mystica555, 2023-07-20:
You closed the issue...What did you do to fix the problem? Future people want to know.

ksaye, 2023-07-20:
I simply gave up.

You are receiving this email because you authored the thread.