botletics/SIM7000-LTE-Shield #177: Confused by which MQTT functions to use for FONA 3G SIM5320

OscarVanL, 2020-04-07:
Hi,

I'm a little confused on what I should use for MQTT on my FONA 3G SIM5320.

I see you've got an MQTT example [here](https://github.com/botletics/SIM7000-LTE-Shield/blob/master/Code/examples/AdafruitIO_MQTT_Demo/AdafruitIO_MQTT_Demo.ino), however this uses the Adafruit MQTT library which does not support the FONA 3G from what I can tell.

Then I see you've also extended the Adafruit FONA library to add MQTT functionality into this library for SIM7000 with an example [here](https://github.com/botletics/SIM7000-LTE-Shield/blob/master/Code/examples/SIM7000_MQTT_Demo/SIM7000_MQTT_Demo.ino).

Are the functions you've written for SIM7000 more likely to work with my FONA 3G?

Additionally, thank you for all the work and documentation you've done on this. I've been pulling my hair out over the official Adafruit FONA libraries which really suck.

botletics, 2020-04-07:
From what I remember, if you use the Botletics SIM7000 library MQTT should work fine on the SIM5320.

OscarVanL, 2020-04-07:
That's a shame. I guess I'm going to need to Frankenstein some of your code then. Thanks anyway :)

OscarVanL, 2020-04-08:
Hi botletics, I have a question about your Adafruit_FONA::mqtt_connect_message function.

I suspect the protocol name length (`mqtt_message[3]` in `mqtt_connect_message`) is being set incorrectly as well as the length calculated in `mqtt_sendPacket` in `Adafruit_FONA::MQTTconnect` when using "MQTT" instead of "MQIsdp" as the protocol name.
This is always being assumed to be 6 long, even if using 'MQTT' as the protocol which is 4 long.

Here's the packet sent by my FONA using AT+CIPSEND:
```

14:04:33.741 -> 	---> AT+CIPSEND=0,48
14:04:34.283 -> 	<--- >
14:04:34.283 -> 	---> 16 44 0 6 77 81 84 84 3 194 0 15 0 15 56 54 55 48 53 55 48 51 49 54 57 53 56 54 49 0 5 111 115 99 97 114 0 8 99 101 110 115 111 114 101 100 48 69 26
14:04:34.451 -> 	<--- OK
```
which becomes Hex 

```
10 2C 00 06 4D 51 54 54 03 C2 00 0F 00 0F 38 36 37 30 35 37 30 33 31 36 39 35 38 36 31 00 05 6F 73 63 61 72 00 08 63 65 6E 73 6F 72 65 64 30 45 1A
```

And now with hex digits replaced by characters where relevant

10 2C **00 06** _M Q T T 3_ C2 00 0F 00 0F 38 36 37 30 35 37 30 33 31 36 39 35 38 36 31 00 05 o s c a r 00 08 c e n s o r e d **30 45** 1A

00 06 = Protocol name length (6). [This should be 00 04 according to heading 3.1.2.1 in the MQTT 3 Protocol specification.](http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Toc398718028)

30 45 = Random garbage appended due to incorrect length calculation in mqtt_sendPacket.

1A = <^Z> (end of packet character added by mqtt_sendPacket)

The resolution would be to have a conditional statement that sets mqtt_message[3] to 6 if using MQIsdp or 4 if using MQTT (using `protocol_length`), and to use 
`14+strlen(protocol)+strlen(clientID)+strlen(username)+strlen(password)` instead of 
`20+strlen(clientID)+strlen(username)+strlen(password)`

botletics, 2020-04-20:
Sorry it took so long to reply but thanks for the catch! You're right, I think
`mqtt_message[3] = 6;`
just needs to be changed to
`mqtt_message[3] = protocol_length;`

OscarVanL, 2020-05-21:
Apologies for not being clearer. The `20+strlen(clientID)+...` line is in `Adafruit_FONA::MQTTconnect` in `Adafruit_FONA.cpp`.

Line 2387 saying 

`if (! mqtt_sendPacket(mqtt_message, 20+strlen(clientID)+strlen(username)+strlen(password))) return false;` 

should be changed to 

`if (! mqtt_sendPacket(mqtt_message, 14+strlen(protocol)+strlen(clientID)+strlen(username)+strlen(password))) return false;`

botletics, 2020-05-21:
Ahh yes you're right, thanks. Fixed it.

closed 2020-05-21 by botletics