botletics/SIM7000-LTE-Shield #178: Firmware update - Waiting switch To Download Mode

kristo2301, 2020-04-13:
Hello, 

During update of firmware to 7000G i get following request but no idea how to switch to download mode, please fupport

[image]

botletics, 2020-04-13:
It should do it by itself. Make sure you have a LiPo battery connected.

botletics, 2020-04-13:
You could also try [these instructions](https://github.com/botletics/SIM7000-LTE-Shield/blob/master/SIM7000%20Documentation/Firmware%20Updater%20Tool/How%20to%20Update%20SIM7000%20FW%20version%20with%20QPST.pdf)

kristo2301, 2020-04-13:
Hi @botletics, 
Thanks for quick feedback,  issue is same in QPTS i get error to start sahara download i think do to HW i have SIM7000G with ESP32 on one board LILY Go -> https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G

Humancell, 2020-04-13:
I think I pressed the "reset" button on mine to get it to download mode ...

kristo2301, 2020-04-13:
@Humancell, 

Thanks for tip but i in my case reset is not helping. I make power off and on to go next step but still not able to finish updating :(

botletics, 2020-04-13:
Make sure you have nothing else connected to it except a LiPo battery.

kristo2301, 2020-04-13:
I have esp32 connected on same pcb and no way to disconnect in simple way.

Humancell, 2020-04-14:
I realize now that I looked at the LilyGo board.

Have you reached out to LilyGo to ask them how you get the board into download mode?  You purchased the board form them, and it seems they must have some steps/instructions on how to update the module firmware with their board and PCB design?

stbrie, 2022-06-12:
@DevinCarpenter - Your solution worked for me.  This information should be included in the firmware update instructions wiki page.

DevinCarpenter, 2022-06-19:
@stbrie Great! Glad to hear that it is working for other people.

@botletics Do you think this should also be updated in the Wiki? Perhaps I can fork this and submit a wiki update.

botletics, 2022-06-19:
I put a link to this in the wiki already, thanks!