botletics/SIM7000-LTE-Shield #181: USB not waking up

rgillan, 2020-04-24:
Hello, I have a recently purchased v6 7000G module and trying to get connectivity up via the USB.  We are running it on an anti static bench, with a solid 5V supply (to +5/GND) and correctly polarised 5000mAh LiPo connected.  The unit wakes up as far as the green power light goes, with the blue netlight flashing on ~1s cadence (indicating that it's not locked LTE).  We have tried with and without SIM but no difference.  We have tried with a Windows machine, a RPi3 and a NanoPi Neo2 (our final target).  If we plug a simple serial/USB converter into either linux machine lsusb shows the device, and there's a ttyUSB0 port in /dev.  When we plug in the USB from the shield we get nothing.  We have verified that the LTE PWR pin also starts/stops the shield (again, the LEDs light).  We must be missing something very basic and any help gladly appreciated.

botletics, 2020-04-24:
Hi there, please make sure you've read [this page] (https://github.com/botletics/SIM7000-LTE-Shield/wiki/USB-Drivers) to install the USB drivers, then see [this page] (https://github.com/botletics/SIM7000-LTE-Shield/wiki/Connecting-to-Network). For the SIM7000G you will likely need to use the commands to narrow down the band selection to get it to connect.

rgillan, 2020-04-24:
Thanks for the prompt response.  Yes we have already gone through the USB driver process on Ubuntu/Armbian (which already has the Qualcomm driver module), Raspberry Pi and Windows 7.  When the device is plugged into the USB port it does not even register with lsusb or the Device Manager.  We have left the charger on overnight and indeed we now have the power led, the ready/charged led and the flashing network led (but not connected, that's not our issue, it's the basic USB waking up).  The only connections we have to the unit are a 3A/5V and ground connection from a power supply; the LiPo and the USB cable.  We have validated that there is a 5V signal on the V_BUS signal coming in so the 7000 should be detecting that the USB is active, it's just appearing not to.  As there's no circuitry between the micro USB and the 7000 the only thing we can think of is that the 7000 is faulty unless we're missing something totally obvious.  Suggestions gladly accepted.

botletics, 2020-04-24:
Please check to make sure the LiPo battery has proper polarity. The red wire should be on the left side of the JST connector when you plug it into the board. Some manufacturers flip the polarity.

rgillan, 2020-04-24:
Yes, we've done that, and checked the voltages/polarity with a DMM (we did change the polarity on the LiPo before we first connected it).  The resultant voltage on VIN is measuring within the operating range of the 7000.  We're a bit stumped.

botletics, 2020-04-24:
Hmmm yea if the voltage is right and the green "PWR" LED is on (with blue NET light as well) then it should show up in device manager as long as your USB cable isn't a power-only cable.

rgillan, 2020-04-24:
Hmm, we've changed USB cables a number of times but will pull out a new one we know is not just power and see (now that would be embarrassing ;) ).  Really appreciate your responsiveness, thank you

rgillan, 2020-04-24:
Well Timothy there's times in your life we're you miss really simple things, and this was one of them.  Turns out the 6 or 7 cables we tested are power only.  Being an Apple shop this whole microUSB thing is only to power devices.  We have pulled a new cable out of a bag that we know is a data cable and voila.  On the Armbian target (which already has the Qualcomm driver) all the ttyUSBx ports immediately pop up (sigh).  Thanks for your help and we'll let you know if there are any issues (but doubt it now)

closed 2020-04-24 by rgillan

botletics, 2020-04-24:
Haha, happens to the best of us... glad you got it working!