botletics/SIM7000-LTE-Shield #183: Eclipse Mosquitto

shutterspark, 2020-05-13:
is there any appetite to add a support for publishing data to Mosquitto? Been at it for days with no luck. Seems like every time I find libraries that support Mosquitto they conflict with the modified Fona Libraries to use the modem.

botletics, 2020-05-13:
It already can publish to CloudMQTT and other MQTT brokers so you should be able to just change the variables in the MQTT example sketch and it should work.

shutterspark, 2020-05-13:
I have tried that relentlessly. Everytime I get the following....

14:53:48.336 -> 	---> AT+SMSTATE?
14:53:48.370 -> 	<--- +SMSTATE: 0
14:53:48.439 -> 	---> AT+SMCONF="URL","xxxxxxxxx","1883"
14:53:48.508 -> 	<--- OK
14:53:48.543 -> 	---> AT+SMCONF="USERNAME",xxxxxxxxxx
14:53:48.578 -> 	<--- OK
14:53:48.613 -> 	---> AT+SMCONF="PASSWORD",xxxxxxxxx
14:53:48.648 -> 	<--- OK
14:53:48.717 -> 	---> AT+SMCONF="TOPIC","xxxxxx"
14:53:48.752 -> 	<--- OK
14:53:48.786 -> 	---> AT+SMCONF="MESSAGE","HELLO"
14:53:48.821 -> 	<--- OK
14:53:48.821 -> Connecting to MQTT broker...
14:53:48.855 -> 	---> AT+SMCONN
14:53:53.885 -> 	<--- 
14:53:53.885 -> Failed to connect to broker!

I am wondering if Mosquitto requires a different aunthentication process? I was hoping someone on here has had luck publishing to Mosquitto. 

Also curious about using MQTTS and sending SSL certs.

shutterspark, 2020-05-14:
Disregard, I got it to publish and am extremely happy with your libraries! Thanks a ton! Now if anyone has connected to Mosquitto using SSL certs, i would like to know how to do that.

jeffryr, 2020-05-18:
How did you ultimately get it to work?  The following arduino code returns an error after trying to publish.

```
#include "Adafruit_FONA.h" // https://github.com/botletics/SIM7000-LTE-Shield/tree/master/Code
#define SIMCOM_7000 // SIM7000A/C/E/G
#define FONA_TX 16 // ESP32 hardware serial RX2 (GPIO16)
#define FONA_RX 17 // ESP32 hardware serial TX2 (GPIO17)
#include <HardwareSerial.h>
HardwareSerial fonaSS(1);
Adafruit_FONA_LTE fona = Adafruit_FONA_LTE();
fonaSS.begin(115200, SERIAL_8N1, FONA_TX, FONA_RX); // baud rate, protocol, ESP32 RX pin, ESP32 TX pin
fonaSS.println("AT+IPR=9600"); // Set baud rate
fonaSS.begin(9600, SERIAL_8N1, FONA_TX, FONA_RX); // Switch to 9600
if (!fona.begin(fonaSS)) {
  Serial.println("Couldn't find FONA");
  delay(2147483647);
}
fona.setFunctionality(1); // AT+CFUN=1
fona.setNetworkSettings(F("hologram")); // For Hologram SIM card
fona.setPreferredMode(38); // Use LTE only, not 2G
fona.setPreferredLTEMode(1); // Use LTE CAT-M only, not NB-IoT
fona.setOperatingBand("CAT-M", 12); // AT&T uses band 12
if (!fona.enableGPRS(true)) {
  Serial.println("Couldn't start GPRS");
  delay(2147483647);
}

if (! fona.MQTT_connectionStatus()) {
    fona.MQTT_setParameter("URL", mqttserver, 1883);
    fona.MQTT_setParameter("USERNAME", mqttuser);
    fona.MQTT_setParameter("PASSWORD", mqttpassword);
    if (! fona.MQTT_connect(true)) {
        Serial.println("Couldn't publish");
        delay(2147483647);
    } else {
        Serial.println("Great success!");
}

```
This is using a SIM7000G module with a Hologram SIM wired to an ESP32. If I connect to the same server using the same parameters via mosquitto_sub it works.

jeffryr, 2020-05-18:
OK, got it working after adding the following after the fona.enableGPRS(true) call, helps to look more closely at the example code :-)

```
    if (!fona.wirelessConnStatus()) {
      while (!fona.openWirelessConnection(true)) {
        Serial.println(F("Failed to enable connection, retrying..."));
        delay(2000); // Retry every 2s
      }
      Serial.println(F("Enabled data!"));
    }
    else {
      Serial.println(F("Data already enabled!"));
    }
```
After successfully publishing, the code crashed with a "Stack smashing protect failure!" in Adafruit_FONA_LTE::MQTT_publish.  This is because the cmdStr buffer in that function was too short for the topic I was passing which was "devkitcv4c/Output/Status".  Not sure what the maximum length should be for the topic, didn't see anything in https://simcom.ee/documents/SIM7000x/SIM7000%20Series_MQTT_Application%20Note_V1.00.pdf other than the 512 byte limit on content length so I changed the size of cmdStr from 40 to 128 bytes and worked around the problem.

OscarVanL, 2020-05-21:
I don't think it's relevant to your issue, but maybe this will help others who search for this problem in the future.

One thing that caused problems connecting to the MQTT Broker for me was using "MQTT" as the protocol in the MQTTconnect function. Using "MQIsdp" as the protocol instead fixed issues with Eclipse Mosquitto.

jason-a-alexander, 2022-03-02:
@OscarVanL I know this is the wayback machine, but where does that get passed.  I'm having some issues with a SIM7000A when I change topics and found this when I was looking for answers.