botletics/SIM7000-LTE-Shield #184: FTP upload image

philhuen, 2020-05-16:
Fail to use image upload, with following error.
invalid conversion from 'const char*' to 'char' [-fpermissive]
Please see if you may help to update the FTP demo

  // Now the really cool part! We're going to upload
  // a picture to the server using the extended PUT
  // method (auto-detected inside the FTP_PUT method
  // based on the content size
  // NOTE: Haven't tested extended PUT method yet because
  // SIM7000G firmware does not support it for some reason...
  
  size_t fileSize;
  char * uploadContent = readFromFile("small-img.jpg", &fileSize);
  Serial.print("File size: "); 
  Serial.print(fileSize); 
  Serial.println(F(" bytes"));
  // Upload picture via FTP
  if (!fona.FTP_PUT("small-img.jpg", "/", uploadContent, fileSize)) { // File name, file path, content, content length
    Serial.println(F("Failed to upload!"));
  }

The readFromFile function is not pointer function:
// Read the contents of a file in the SD card
char readFromFile(const char fileName, size_t * fileSize) {
  char contentBuff[250];
  
  myFile = SPIFFS.open(fileName);
  *fileSize = myFile.size();
  
  if (myFile) {
    // Read from the file until there's nothing else in it
    while (myFile.available()) {
      strcat(contentBuff, myFile.read());
    }
    myFile.close(); // Close the file; only 1 can be open at a time
  }
  else {
    Serial.println("Error opening file!");
  }

  return contentBuff;
}

botletics, 2020-05-17:
Sorry, I haven't had the time to update the FTP features.

danielehrhardt, 2020-10-31:
How huge can a File be to Upload?