botletics/SIM7000-LTE-Shield #185: SIM7000E does not respond AT commands

pdpj, 2020-05-26:
My project uses **SIM7000E** connected to **PIC24FJ128GA204** and so that AT commands are send to and reveice from SIM7000E using  UART. When "AT" is sent the module normally reponds with "OK", but when any other command is sent it only send as answer a copy of the command itselfs. For example, if "AT+CGMR" is sent the received anwser is "AT+CGMR". I tested with many commands and the same happens. What might be this problem?  Is there any special configuration to be done before using general commands?

botletics, 2020-05-27:
Are you using the Botletics SIM7000 board or some other variation? There shouldn't be any special setup required.

pdpj, 2020-06-16:
I'm using the Botletics SIM7000. And fortunately I solved the problem. The situation was happening because I was sending the commands in sequencie but too fast, not giving enough time so that the module could answer. After inserting some delays it is working normally. 

SIM7000 can be powered on by pulling the PWRKEY(LATAbits.LATA10) pin to ground, as the datasheet says.

```
void turnONsim7000 (void) 
{   
    LATAbits.LATA7=1;
    __delay_ms(4000);

    LATAbits.LATA10=1;
    __delay_ms(1000);

    LATAbits.LATA10=0;
    __delay_ms(1000);

    LATAbits.LATA10=1;
     __delay_ms(5000);
}

sendAtCommand("AT\r");
    __delay_ms(100);
```
Thank you for trying to help me!

closed 2020-06-16 by botletics