botletics/SIM7000-LTE-Shield #188: fona.getNetworkStatus() not working after fona.enableSleepMode(true)

hardastyle, 2020-06-09:
After enabling Sleep Mode, it seems like the function getNetworkStatus does not return a valid value.
However, running the command by itself does work.
---> AT+CGREG?
<--- +CGREG: 0,5

My code to check whether data is on:
`uint8_t n = fona.getNetworkStatus();`
`if ((n != 1) && (n != 5))`
`  Serial.println("Connected");
`

botletics, 2020-06-09:
Really? What is it returning?

hardastyle, 2020-06-09:
So I set up a couple of buttons right after the Fona initializes. I use your SMS_response example.
It's like the fona port gets disconnected or corrupted or something, but the strange thing is I can still send commands with fona.sendCheckReply(message, F("OK")).
Let me know if this helps. I, a ME myself, also wanted to add that your work is amazing.

Button 1:

>   uint8_t n = fona.getNetworkStatus();
>   Serial.print("right before sleepmode(true): ");
>   Serial.println(n);
>   fona.enableSleepMode(true);
>    n = fona.getNetworkStatus();
>   Serial.print("right after sleepmode(true): ");
>   Serial.println(n);

Button 2:

>  uint8_t n = fona.getNetworkStatus();
>   Serial.print("right before sleepmode(false): ");
>   Serial.println(n);
>   fona.enableSleepMode(false);
>    n = fona.getNetworkStatus();
>   Serial.print("right after sleepmode(false): ");
>   Serial.println(n);

Serial Monitor:

> 15:40:21.328 -> FONA Ready
>
>
> 15:40:51.112 -> 	---> AT+CGREG?
> 15:40:51.146 -> 	<--- +CGREG: 0,5
> 15:40:51.146 -> right before sleepmode(true): 5
> 15:40:51.214 -> 	---> AT+CSCLK=1
> 15:40:51.214 -> 	<--- OK
> 15:40:51.247 -> 	---> AT+CGREG?
> 15:40:51.282 -> 	<--- +CGREG: 0,5
> 15:40:51.316 -> right after sleepmode(true): 5
>
>
> 15:41:00.838 -> 	---> AT+CGREG?
> 15:41:01.384 -> 	<--- 
> 15:41:01.384 -> right before sleepmode(false): 0
> 15:41:01.417 -> 	---> AT+CSCLK=0
> 15:41:01.927 -> 	<--- 
> 15:41:01.962 -> 	---> AT+CGREG?
> 15:41:02.506 -> 	<--- 
> 15:41:02.506 -> right after sleepmode(false): 0

botletics, 2020-06-09:
Oh, that looks like a power issue. Please check if you have a 3.7V LiPo battery of proper polarity (if you're using the Botletics board the red wire of the battery should be on the left side of the connector when you plug it into the board)