botletics/SIM7000-LTE-Shield #190: GPS location not being found

Alfredfoby, 2020-06-30:
When I use the code given for the IoT and comment out most of the temperature sensor code since it wasn't working for me I get stuck in the while loop that prints "Failed to get gps location" I'm talking about this loop below
while (!fona.getGPS(&latitude, &longitude, &speed_kph, &heading, &altitude)) {
Serial.println(F("Failed to get GPS location, retrying..."));
delay(2000); // Retry every 2s
}

I decided that maybe it was an error and removed the not sign "!" and then it goes through the loop

to next part of the code, the problem now is that I get the wrong latitude and longitude and I also don't get any data for the heading. The longitude and latitude show that I am in Kenya when in reality I am in the United States.

I hope someone can help me.

botletics, 2020-07-01:
Hi, please post a serial monitor log so we can debug, thanks.

Alfredfoby, 2020-07-01:
I have attached the log below, I dont seem to see anything wrong but its just not working. 

[image]
[image]

osy93, 2020-07-02:
Hi Tim, first of all thanks a lot for all the hard work put in this project - I just got the shield and its awesome !

I have the same issue above (my location is in Dubai, UAE), would really appreciate your input !

Regards,
Osama

botletics, 2020-07-02:
Sometimes it takes a couple minutes to get a fix on GPS. How long did you wait?

Alfredfoby, 2020-07-02:
I waited for an hour at some point and still couldn't find a location.

Alfredfoby, 2020-07-02:
I was hoping it had to do with some of the pins that weren't enabled and were commented out in the sketch

botletics, 2020-07-02:
No, as long as GPS was enabled you should get a fix on GPS location eventually. However, inside some buildings there may not be a good signal. If power consumption is not a concern at all you could try soldering the solder jumper labeled "ACTV" on the board (for v6 boards only) and attach an active GPS antenna.

Alfredfoby, 2020-07-02:
It is finally working, All i had to do was taking the arduino for a ride haha, i drove with the modem and arduino for some time and it finally started working. I was just in an area where the gps wasn't working for some reason.

Thank you very much for everything.

closed 2020-07-02 by botletics

ishdemon, 2021-03-30:
As long as you are getting reply from AT+CGNSINF, its working fine. Mine just doesn't reply at all, stuck in a loop :

```
AT+CGNSINF
AT+CGNSINF
AT+CGNSINF
AT+CGNSINF
AT+CGNSINF
AT+CGNSINF
AT+CGNSINF
```
was searching through issues if someone also had this problem. Sometimes after reset it works fine for a while.