botletics/SIM7000-LTE-Shield #192: can't turn on cellular data or gps

andrewhong5297, 2020-07-06:
Using botletics 7000A and hologram sim card, after running LTE_demo the G and O commands don't work for me. I can read network time and signal strength though. Any idea what might be going wrong? 

Both green lights are working (is there a way to make these lights dimmer?), and the net light is blinking blue.

botletics, 2020-07-06:
Sounds like you don't have a LiPo battery attached and it's probably powering off when you try to enable those functions.

andrewhong5297, 2020-07-06:
forgot to mention I have a lipo 3.7v plugged in - is there some indicator to tell if it is on or not?

botletics, 2020-07-06:
Please check the polarity of the LiPo battery wires. The red wire should be on the left side of the connector when you plug it into the board.

The green "PWR" LED indicates whether or not the SIM7000 is powered on.

andrewhong5297, 2020-07-06:
Here's my setup, I cant plug the lipo in any other way.

botletics, 2020-07-06:
Yea that's the problem. Your battery polarity is backwards. Please order a LiPo battery with a minimum of 500mAH capacity, preferably 1000mAH+, from either Adafruit or Sparkfun (can be found on Amazon as well). There are many knock-offs and some have reversed polarity. Your battery also looks pretty small.

closed 2020-07-06 by botletics

andrewhong5297, 2020-07-08:
Can you find an amazon link with correct batteries? I've ordered three times already but its always either the head is slightly off or polarity is wrong. Would greatly appreciate it!

andrewhong5297, 2020-07-09:
for example, they all seem to be like this https://www.amazon.com/YDL-1700mAh-battery-Rechargeable-Connector/dp/B07BTRTR9R/ref=sr_1_3?dchild=1&keywords=2500mah+lipo+battery+3.7v+reverse&qid=1594254091&sr=8-3

botletics, 2020-07-09:
I know [this one will work](https://www.amazon.com/dp/B00J2QET64?ref=ppx_pop_mob_ap_share)