botletics/SIM7000-LTE-Shield #196: SIM7000E crashing over and over

semorem, 2020-08-28:
Hi there, i bought a module SIM7000E and it seems to be crashing all the time, it looks like it is executing over and over "AT+CFUN=1", and its doing this because the "AT+CEDUMP=0", i checked if it was actually crashing by changing to AT+CEDUMP=1 and stays frozen. I only wanna know why is crashing, or whats the problem, cause i have no clue. The strangest command that i had execute was AT+CFUN=6,0; but the rest were just regular configure commands. 
Oh, and if anyone knows how to make a factory reset i would be grateful, because i saw that ATZ was deleted on the v1.02

Sorry for my english, and thanks in advance!

botletics, 2020-08-28:
What hardware are you using? Does it have a sufficient power supply? Try "AT&F0" to factory reset.

semorem, 2020-08-31:
I have tried several ways. I tried connecting just the module (SIM7000E) via USB to the PC, i tried connecting the module to a Raspberry Pi 3 B+ powering it via USB to the PC and to a regular plug and accessing it through putty using the IP and with the OS Raspbian. 
I also tried the command AT&F0 and it looks like its doing nothing, i see the next info when i used it:

Autobaud support: (9600,19200,38400,57600,115200).

OK

But nothing seems to change.
Sorry if it's poor info but i really don't know what to do.

aemoreno1982, 2020-08-31:
You haven't yet specified which _hardware_ you're using.

semorem, 2020-09-01:
Sorry, my bad, i'm usin a SIM7000E NB-IoT HAT compatible with Raspberry Pi 2B/3B/Zero/Zero W

cynicaljimbo, 2020-09-12:
You might try upgrading the firmware per the instructions on GitHub:  https://github.com/botletics/SIM7000-LTE-Shield/wiki/Updating-Firmware  if it's not too braindead.  There are a lot of issues with the earlier f/w versions.  There's also no way that I know of that you can completely zap it back to factory default- some f/w versions support AT&F0 (I think that's the right command) and some don't, but I've never seen it work correctly even when "supported".

semorem, 2020-09-17:
I already have update the firmware, and now im working with qmicli in linux. 
The modem is able, theoretically, to work with GSM (2G) and Cat-M and NB-IoT (4G). With GSM im having no trouble, with Cat-M im not sur if the SIM Card is capable of work with it, but with NB-IoT i know for sure that it can. 
When i set the modem to NB-IoT mode, it works for a few seconds, i can even send a Ping Request and it works just fine, but a few seconds later it just crash for no apparent reason. 
If anyone could help me i would be really grateful

cynicaljimbo, 2020-09-17:
Here in the States as far as I can tell the SIM doesn't care about the service, just the provider and the modem and what you've subscribed to- but let's ignore that for now since you have NB-IoT working.  Are you using a battery in conjunction with the UNO USB power?  According to the SIMCOM documentation, the 7000 will draw significant current spikes that the USB may not be able to supply- if you don't have a battery, the shield needs a capacitor to buffer the load which the Bioletics shield doesn't appear to have (mine does, it's not a real Bioletics).

semorem, 2020-09-18:
Im supplying the Raspberry with 5 V and the HAT also with an external power supply (not using UNO) It has no sense that the cause is a power issue, because GSM demands more amps than the LTE mode.

cynicaljimbo, 2020-09-18:
Look at http://www.shenzhen2u.com/doc/Module/SIM7000C/SIM7000%20Hardware%20Design_V1.01.pdf : Per Table 6, GSM drain spikes 2A but LTE is still 600mA.  I know you said GSM works, but...   Regardless, I've had a lot of trouble with the SIM7000 module, and so have other people.  Maybe we're getting counterfeit modules?

I'm testing the 7020G right now, I haven't finished, but it's a lot better so far.